ChHarding/TouchTerrain_for_CAGEO #4: How do I get started?

r-barnes, 2017-01-16:
The `ReadMe.md` file does not give users a very good idea about which of 7 python files to run, or what they do. Configuration is apparently necessary, but not well-described by the readme.

ChHarding, 2017-01-23:
Richard,

I’ve worked up an installation workflow for the stand alone version. It’s still somewhat convoluted but the good news is that it can be run with any Google account, rather than a service account as the server version does. 

https://docs.google.com/document/d/1kgdLu_CuHRcopOV7ezB1A1h-aVo2gwbUNtbvmFlR0tg/edit?usp=sharing <https://docs.google.com/document/d/1kgdLu_CuHRcopOV7ezB1A1h-aVo2gwbUNtbvmFlR0tg/edit?usp=sharing>

I’d be grateful if you could try it and tell me if this works for you. If it does, I’ll add a link to it to the ReadMe file on Github.

Cheers

Chris

r-barnes, 2017-01-27:
Chris,
I'll try to take a look sometime in the next couple of days.

Cheers,
Richard

r-barnes, 2017-02-05:
I'm waiting for EE approval from Google now.

Near the top of the document, you might want to add a line about how Linux users can install pip with `sudo apt install python-pip`.

Relatedly, Linux users will need to run `sudo apt install libssl-dev openssl` before they can install Python's cryptographic libraries.

It would probably be better to replace some of these images with text to facilitate copy and pasting by the user, for instance, the one following the line "Now you can test the ee python API. Open a python interactive shell and type this, which should print a dictionary with metadata about the SRTM90 layer:".

ChHarding, 2017-02-06:
> I'm waiting for EE approval from Google now.
> 
Interesting. I had hoped that my workflow would use the everyday Google account for authentication. But, I was already on the EE list when I wrote this, so that’s very good to know! At what point did it want you to get signed up with EE? Or, what didn’t work and how did you figure out that you needed to signup? I’d like to put that into the instructions.
> Near the top of the document, you might want to add a line about how Linux users can install pip with sudo apt install python-pip.
> 
> Relatedly, Linux users will need to run sudo apt install libssl-dev openssl before they can install Python's cryptographic libraries.
> 
Done
> It would probably be better to replace some of these images with text to facilitate copy and pasting by the user, for instance, the one following the line "Now you can test the ee python API. Open a [...]
Done. 

I’ve also added a requirement for the vector module. It’s actually not yet in use in the version that’s on github right now but I’ve been tinkering with the code and it will be needed in the future.

Thanks for testing!

r-barnes, 2017-02-06:
Without an approved EE account, the following happens:

```
root@ubuntu-512mb-nyc3-01:~# python -c "import ee; ee.Initialize()"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/ee/__init__.py", line 93, in Initialize
    ApiFunction.initialize()
  File "/usr/local/lib/python2.7/dist-packages/ee/apifunction.py", line 151, in initialize
    signatures = data.getAlgorithms()
  File "/usr/local/lib/python2.7/dist-packages/ee/data.py", line 398, in getAlgorithms
    return send_('/algorithms', {}, 'GET')
  File "/usr/local/lib/python2.7/dist-packages/ee/data.py", line 738, in send_
    response.status)
ee.ee_exception.EEException: Server returned HTTP code: 404
```

Now that my EE account is approved (it took about a day), this happens:

```
root@ubuntu-512mb-nyc3-01:~# python -c "import ee; ee.Initialize()"
root@ubuntu-512mb-nyc3-01:~# 
```

which indicates success, I suppose.

r-barnes, 2017-02-06:
> I’ve also added a requirement for the vector module. It’s actually not yet in use in the version that’s on github right now but I’ve been tinkering with the code and it will be needed in the future.

I actually had to install `vectors` to get things working when I tested the `TouchTerrain_standalone.py <your_config_file.json>` directive in the readme.

Do you know how to tag commits as releases on Github? I see that you have 0.10, 0.11, and 0.12 listed in the readme, but I'd also expect to see these as tags/releases. This has the benefit that once you reach a point where the code is stable, such as for publishing in CAGEO, you can tag that commit and it will be easy for people to find. Another way to manage the code is by doing all the development on a separate branch and having each merge to master be a new tagged commit.

Either way, you'll want to do something so that people reading your manuscript don't find unintentionally broken code (and, conversely, can easily find working code) when they come to your Github.

r-barnes, 2017-02-06:
Okay, standalone produced an STL file for me! (I really appreciate that the config options get printed to the command line and are included in the ZIP file.)

Two additional thoughts:
 * It would be nice to have a brief section on starting the webserver. A guide to configuring nGINX (beyond listing what port your app defaults to) is not necessary.
 * You might consider reducing the number of files in the directory by relocating utility scripts such as "InMemoryZip.py" to a subdirectory. That way users are presented with limited options to choose the wrong script, &c.

ChHarding, 2017-02-07:
That’s also weird. From my own experiments, I was expecting that a failing python -c "import ee; ee.Initialize()” should tell you to authorize access EE by running

earth engine authenticate

which you didn’t get and so I assume you didn’t try the earth engine authenticate thing. I guess I just point towards the EE signup until I can figure out what the deal is.    
 
> Without an approved EE account, the following happens:
>
> root@ubuntu-512mb-nyc3-01:~# python -c "import ee; ee.Initialize()"
> Traceback (most recent call last):
>   File "<string>", line 1, in [...]

r-barnes, 2017-02-07:
Actually, I did run `earthengine authenticate` a couple of times and still got the 404 message. Which is odd, and probably outside your control.

ChHarding, 2017-02-07:
Nice to see you got a stl file! I can certainly uncluttered the folder as per your suggestions.  Our server guy has a ansiable script that he uses to kickstart out apache sever that could be put on Github as well. I have to admit I’m still fairly new to git. Good idea to use tags, I’ll work on that.  Thanks for the feedback!

r-barnes, 2017-02-08:
No worries! Let me know if there's anything else I can do to help.

ChHarding, 2017-02-14:
I actually have another question on github structuring.  I’m thinking of having two folders server and standalone at the github top level. Both would contain only the files needed for installing and running their version.  But - both folders would have to share four or so .py files. On unix, I could of course simply use soft link them e.g. put the real files in the server folder and link them to the stand alone folder. Is that a recommended strategy with git? Or will that lead to trouble down the road?

r-barnes, 2017-02-14:
I'd actually recommend keeping those common files in a separate directory called `lib/` or something. You could then import the files as described [here](http://stackoverflow.com/a/7506029/752843).

ChHarding, 2017-02-16:
I read up on doing relative imports but ended up confused. I found a solution but it doesn’t strike me as very elegant. Could you have a look? (This is just a test case BTW)

The main file is in the server folder, the common folder has a module the main file needs to import. The __init__.py files are empty

I would have like to do something simple like

from ..common import mod1 

but always ended up getting ValueError: Attempted relative import in non-package.  Or maybe there’s some magic in what the __init__.py file need to contain for this to work?

Cheers  

Chris

r-barnes, 2017-02-16:
I wonder if you could push a branch to Github in which the files are arranged appropriately (in 'standalone', 'server', and 'common' directories, I think)? Once that's done, I can fork and poke around at it.

ChHarding, 2017-02-16:
Done. The standalone version now main now works with this:

maybe that’s OK but if you can figure out how to to the ..common imports instead, that would be great!

Cheers

Chris

closed 2017-02-16 by ChHarding

r-barnes, 2017-02-17:
It looks like the current setup might be about as good as we can get here. I've just issued a pull request which cleans up the ReadMe file. Everything else is looking pretty decent.

ChHarding, 2017-02-17:
Very cool! Thank you so much. As you can tell, this is my first github project and having a more experienced person rearrange things into a more standard way is really helpful. I hope you didn’t spend too much time fixing up the ReadMe.md, it looks very nice! 

BTW I was also thinking about just having a central license statement (either at the end of the readme or in a separate file) instead on the top of each python file.

Thanks again!

r-barnes, 2017-02-23:
I don't know if it's more standard, but, as an outsider come to the project, I find it easier to understand. Some of it matches the CRAN layout, as I may have mentioned above.

You'll note that editing the `README.md` file has changed the landing page of the Github repo. For this reason, having good content in this file is important for people considering downloading and working with your code.

Having a central license statement is a good idea as it will make the project searchable by license and makes it clear to a user what the license for the project is, rather than just individual files. It should be stored in a file entitled `LICENSE.txt` or `LICENSE.md`. Check out [choosealicense.com](https://choosealicense.com/) if you need help choosing a license.