ChHarding/TouchTerrain_for_CAGEO #7: Server Setup

phubbard91, 2018-08-29:
Thanks for the work on this! It is a very cool project. I'm attempting to setup the server side and have been able to get through all of the authentication, however, when I run the python script to setup the server, it finishes with no server set up. I think I am missing the step on editing the ansible script, but I am not familiar with .yaml formatting or ansible in general. Could someone help me out? Thanks.

ChHarding, 2018-08-29:
Hi!

I’m also not a server guy so I’m redirecting your question to Levi, who set up our ISU server and wrote the ansiable script for it. I’m confused as to what you call  the python script to setup the server. My impression is that running the ansiable script should setup the folders, copy stuff into them, etc. From what I know to actually start the server using systemctl start  httpd.  But hopefully Levi can help you with this.

Also, I’m going to commit a new version today or tomorrow which has one major improvement, so be ready to update to that version soon.

Cheers

Chris

> Thanks for the work on this! It is a very cool project. I'm attempting to setup the server side and have been able to get through all of the authentication, however, when I run the python script to [...]

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

phubbard91, 2018-08-29:
Chris,

Thank you so much for the quick response. I really appreciate it. I'll wait to hear back from Levi and look for the updates over the next few days.

Best,

Patrick

baberlevi, 2018-08-30:
The server_setup_script_example_ansible.yml is probably not very up to date.   It was really just a template that someone could build off of, but probably never directly runnable outside of our environment. I'm updating a stand-alone version here: https://github.com/baberlevi/touchterrain-ansible-deploy/ and will let you know when it's ready.

Also, we run RHEL, so you won't be able to take this as-is and run it on Ubuntu (but you could on CentOS).  I may make a docker image at some point to make it easier for folks to run their own.

phubbard91, 2018-09-05:
Thanks for the update. A docker image would be awesome, but I could probably manage getting it running on CentOS. I'll be checking back periodically for more updates. Thanks again for your time.

ChHarding, 2018-09-05:
You’ve probably seen that there’s a new version out. Sadly I can’t help you with getting the server running. In the mean time, have you played around with the stand alone version? Could that be useful for you?

> 
> Thanks for the update. A docker image would be awesome, but I could probably manage getting it running on CentOS. I'll be checking back periodically for more updates. Thanks again for your time.

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

phubbard91, 2018-09-12:
Thanks Chris, I have been able to get the standalone version to work. I am having a small problem though; when I attempt to save the downloaded geotiff from earth engine, it shows up as empty. I tried using your debugging method in line 400 of TouchTerrainEarthEngine.py to save it. Any advice on how I can save the geotiff locally? 

Thanks,

Patrick

ChHarding, 2018-09-12:
Empty as in 0 bytes?  Or does it not show any elevation values (everything
is white)? If so, how are you looking at the geotiff? As it's a 32-bit
floating point tiff, I'm guessing that most non-GIS viewers won't show you
anything (even GIMP doesn't deal with floating point tiffs ...)
But. I do see this in QGIS (the file is 512kb), so the values are in there:

[image: image.png]

If its 0 byes, look in the log file after it''s done, it should tell you
the number of pixels it wrote via PIL, for me that was 453 x 289 pixels.

Google Earth Engine raster: USGS/NED USGS National Elevation Dataset 1/3
arc-second 'link'
d/l geotiff size: 0.500308990479 Mb
<PIL.TiffImagePlugin.TiffImageFile image mode=F size=453x289 at 0x53D3CF0>

I'm attaching my json file that created the GeoTiff  (test.json), see if
that works for you.

BTW at that point the raster is not a real geotiff anymore, it's just a
tiff. It doesn't have a coordinate system or projection b/c I write it via
PIL, which doesn't write the Geo-specific info into the file header. But,
it could be a proper GeoTiff, if I were to use GDAL instead of PIL. If
that's important for you, let me know.

Cheers

Chris

baberlevi, 2018-09-20:
I finally had a chance to work on the docker container.  I think this will be a lot easier for you to deploy locally.

You can find the instructions & Dockerfile here: https://github.com/ResearchIT/touchterrain_docker

Please try it out & let me know if it works for you.

phubbard91, 2018-09-24:
Thanks for the update. I found that the standalone worked well, but will try the docker when I have some more time. One thing I have noticed: I run out of memory when trying to process large STL files (100 mm tiles at 0.1 mm resolution).  I have tested this on both a 2 GB RAM Ubuntu virtual machine and my 8 GB RAM macbook. Are there any changes to settings that I can make to avoid this?

baberlevi, 2018-09-24:
That sounds right. @ChHarding has changed the lowest resolution to 0.2mm available on the website, and we run the production VM with 24G of RAM.

closed 2019-01-03 by ChHarding