ChHarding/TouchTerrain_for_CAGEO #9: [Errno 28] No space left on device:

ecridlebaugh, 2018-11-24:
Running into this error and unable to create/download STL files.

[Errno 28] No space left on device: '/var/www/html/touchterrain/server/tmp/596314982299_dem.tif'

ChHarding, 2018-11-25:
Sorry about this, I’m guessing the server is out of temp disk space.

Levi - can you fix this? Thanks!

Eric - I’ll ping you once it’s fixed

Cheers

Chris

> 
> Running into this error and unable to create/download STL files.
> 
> [Errno 28] No space left on device: '/var/www/html/touchterrain/server/tmp/596314982299_dem.tif'

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

ChHarding, 2018-11-26:
Maybe there was a large temp file there previously, but there is not now (13G free)

closed 2019-01-03 by ChHarding