ChHarding/TouchTerrain_for_CAGEO #21: Type in Coordinates

DakotaWray2, 2019-10-19:
I would like the ability to type in coordinates for fine-tuning rather than rely on the dragging the box on the map. I will try to implement when I have time and create a pull request. Thanks!

ChHarding, 2019-10-23:
Great, thanks for the offer. I’ve thought about this a bit and there was something in the way it’s now implemented that gave me pause, but now I forgot what that was. Also be aware that this was literally my first multi page piece of JS code and it's probably very “nooby”, so eyeroll/facepalm warning :). It probably reads more like C than actual JS … I still want to keep is as pure JS as possible but I’m aware that many things could be improved anyway.

Looking at the code now (after 3 years), I think the 4 text boxes trlat2, etc. would need to be set to not readonly and have a callback when enter was pressed, either 4 separate callbacks or one that’s fired if one or more of the 4 have been changed. Or, instead of enter callbacks something like a loose focus callback?

  	 
         <input type="text" id="trlat2" name="trlat" maxlength="7" size="7" value="NULL" title="(trlat)" readonly> N
         <input type="text" id="trlon2" name="trlon" maxlength="8" size="8" value="NULL" title="(trlon)" readonly> E (Top right corner)<br>

         <input type="text" id="bllat2" name="bllat" maxlength="7" size="7" value="NULL" title="(bllat)" readonly> N
         <input type="text" id="bllon2" name="bllon" maxlength="8" size="8" value="NULL" title="(bllon)" readonly> E (Lower left corner) <br>
         <br>

In each of those callbacks, the appropriate side of the bounds of the global rectangle has to be changed, similar to this but with set instead of get:

      let b = rectangle.getBounds();
      let trlat = b.getNorthEast().lat();
      let trlon = b.getNorthEast().lng();
      let bllat = b.getSouthWest().lat();
      let bllon = b.getSouthWest().lng();

If you ever get started on this and need help, please ping me!

Cheers

Chris

> 
> I would like the ability to type in coordinates for fine-tuning rather than rely on the dragging the box on the map. I will try to implement when I have time and create a pull request. Thanks!

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

ChHarding, 2020-01-14:
BTW (just thought of this now ...) you can sort of type in your coordinates:
1) position the red box roughly where you want it 
2) in the box for manual settings, overwrite the coordinates for the box corners, eg. "trlat":44.5, "trlon":-101.5, "bllat":37.1, "bllon":-111
tr = top right, bl = bottom left corner
3) the red box won't visually move to these overwrite coordinates but they will be used for processing!

Let me know if that works for you!

ChHarding, 2020-10-17:
This has been added to version 3.0

closed 2020-10-17 by ChHarding