ChHarding/TouchTerrain_for_CAGEO #22: Website is down

alexdresko, 2019-12-26:
Just wanted to point out that the site has been down for a week or two. I miss being able to use touchterrain. Will it be coming back online, or do I need to install it locally?

ChHarding, 2019-12-26:
Alex,

I just today noticed that myself and asked our IT guys about this. As of about 30 min ago its been worked on. I’ll let you know when its up again

> 
> Just wanted to point out that the site has been down for a week or two. I miss being able to use touchterrain. Will it be coming back online, or do I need to install it locally?

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

ChHarding, 2020-01-02:
Alex,

its back up again. Sorry for the outage! Also, I’m curious - what do you use the app for?

Cheers

Chris

> 
> Just wanted to point out that the site has been down for a week or two. I miss being able to use touchterrain. Will it be coming back online, or do I need to install it locally?

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

alexdresko, 2020-01-02:
3d printing for fun... But the only one I currently have planned to do is the area surrounding my wife's childhood home in the mountains. 

Hope the fix wasn't too difficult. Thanks for bringing it back to life!

ChHarding, 2020-01-02:
I don’t think it was tricky, it was just really bad timing with all the holydays and no one around to fix it sooner.

> 
> 3d printing for fun... But the only one I currently have planned to do is the area surrounding my wife's childhood home in the mountains.
> 
> Hope the fix wasn't too difficult. Thanks for bringing it back to life!

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

closed 2020-01-14 by ChHarding