ChHarding/TouchTerrain_for_CAGEO #24: ValueError: cannot set WRITEABLE flag to True of this array

ljverge, 2019-12-29:
**Any thoughts on why I am getting "ValueError: cannot set WRITEABLE flag to True of this array"?**

ValueError                                Traceback (most recent call last)
<ipython-input-3-d2e750ee24f6> in <module>
      4 # This may take some time! You'll see In[*] and some log messages (will also be in the logfile inside the zip)
      5 # You may see some red stuff with 10%, etc. - don't worry, that's normal
----> 6 totalsize, full_zip_file_name = TouchTerrain.get_zipped_tiles(**args) # all args are in a dict
      7 print("\nDONE!\n\nCreated zip file", full_zip_file_name,  "%.2f" % totalsize, "Mb")
      8 

~\Downloads\TouchTerrain_for_CAGEO-master\common\TouchTerrainEarthEngine.py in get_zipped_tiles(DEM_name, trlat, trlon, bllat, bllon, importedDEM, printres, ntilesx, ntilesy, tilewidth, basethick, zscale, fileformat, tile_centered, CPU_cores_to_use, max_cells_for_memory_only, temp_folder, zip_file_name, no_bottom, bottom_image, ignore_leq, unprojected, only, original_query_string, no_normals, projection, **otherargs)
    780             cell_size_m *= scale_factor
    781             pr(" ",npim.shape[::-1], adjusted_print3D_resolution, "mm ", cell_size_m, "m ", numpy.nanmin(npim), "-", numpy.nanmax(npim), "m")
--> 782             npim.flags.writeable = True # not sure why it's not writeable after the PIL interpolation but this fixes it
    783 
    784             if adjusted_print3D_resolution != print3D_resolution:

**ValueError: cannot set WRITEABLE flag to True of this array**

ChHarding, 2019-12-30:
Hi!

Glad to see you solved your gdal install issue!  How did you solve it?

As to the writeable flag issue, I sadly don’t understand this either. After I implemented the resampling I noticed that the resampled array was not writeable (which is needed later), so I set this flag to True manually but without really understanding why it wasn’t set in the first place.  I suspect that the reason it doesn’t work for you is that you may have  a different (newer) version of bumpy, so please try this:

1) tell me what version of numpy you have installed and on what OS
2) comment out npim.flags.writeable = True and see if you get errors later when npim gets changed.  If it works the I will just upgrade to your version of numpy. If not I will have to re-investigate why the re-sampled array is strange.

> 
> Any thoughts on why I am getting "ValueError: cannot set WRITEABLE flag to True of this array"?
> 
> ValueError Traceback (most recent call last)
> in 
> 4 # This may take some time! You'll see In[*] and some log messages (will also be in the logfile inside the zip)
> 5 # You may see some red stuff with 10%, etc. - don't worry, that's normal
> ----> 6 totalsize, full_zip_file_name = TouchTerrain.get_zipped_tiles(**args) # all args are in a dict
> 7 print("\nDONE!\n\nCreated zip file", full_zip_file_name, "%.2f" % totalsize, "Mb")
> 8
> 
> ~\Downloads\TouchTerrain_for_CAGEO-master\common\TouchTerrainEarthEngine.py in get_zipped_tiles(DEM_name, trlat, trlon, bllat, bllon, importedDEM, printres, ntilesx, ntilesy, tilewidth, basethick, zscale, fileformat, tile_centered, CPU_cores_to_use, max_cells_for_memory_only, temp_folder, zip_file_name, no_bottom, bottom_image, ignore_leq, unprojected, only, original_query_string, no_normals, projection, **otherargs)
> 780 cell_size_m *= scale_factor
> 781 pr(" ",npim.shape[::-1], adjusted_print3D_resolution, "mm ", cell_size_m, "m ", numpy.nanmin(npim), "-", numpy.nanmax(npim), "m")
> --> 782 npim.flags.writeable = True # not sure why it's not writeable after the PIL interpolation but this fixes it
> 783
> 784 if adjusted_print3D_resolution != print3D_resolution:
> 
> ValueError: cannot set WRITEABLE flag to True of this array

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

ljverge, 2019-12-31:
**Chris,** 

**It turns out I was just trying to install the incorrect version of gdal. I was unfamiliar with .whl installation. All good with that now.**

**I am running numpy 1.16.5 on windows 10.** 

**When I comment out "npim.flags.writeable = True" I get another error further down because it's not writeable**

~\Downloads\TouchTerrain_for_CAGEO-master\common\TouchTerrainEarthEngine.py in get_zipped_tiles(DEM_name, trlat, trlon, bllat, bllon, importedDEM, printres, ntilesx, ntilesy, tilewidth, basethick, zscale, fileformat, tile_centered, CPU_cores_to_use, max_cells_for_memory_only, temp_folder, zip_file_name, no_bottom, bottom_image, ignore_leq, unprojected, only, original_query_string, no_normals, projection, **otherargs)
    838         # apply z-scaling
    839         if zscale != 1.0:
--> 840             npim *= zscale
    841             #print "elev min/max after x%.2f z-scaling: %.2f to %.2f" % (zscale, numpy.nanmin(npim), numpy.nanmax(npim))
    842 

ValueError: output array is read-only

ChHarding, 2020-01-02:
I’m at a slightly older numpy version (1.16.2) but after reading up on numpy’s flags, I realized that it’s an issue with the way that I resize the array using PIL images. Not sure why the write flag thing changed for your version but at least now I have a solution rather than a hack :)

The fix is simple, just insert this line in resampleDEM() before return a (should be TouchTerrainEarthEngine.py Line 240) :

a = numpy.copy(a)

So change this:

    if has_nan:  # swap NaN back in
        a = numpy.where(a < nanmin, numpy.nan, a)
        #print "resample4 min/max : %.2f to %.2f" % (numpy.nanmin(a), numpy.nanmax(a))
        
    return a

to this:

    if has_nan:  # swap NaN back in
        a = numpy.where(a < nanmin, numpy.nan, a)
        #print "resample4 min/max : %.2f to %.2f" % (numpy.nanmin(a), numpy.nanmax(a))
        
    a = numpy.copy(a)
    return a

Keep npim.flags.writeable = True uncommented or delete it.

See if that works, If it does I’ll put it into the Beta version first (as I’m doing some other upgrades also) and once that’s been tested, I’ll copy all the changes into Master.

> Chris,
>
> It turns out I was just trying to install the incorrect version of gdal. I was unfamiliar with .whl installation. All good with that now.
>
> I am running numpy 1.16.5 on windows 10.
>
> When I [...]

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

ljverge, 2020-01-03:
**That worked for me! With that I end up with "pyramid_tile_1_1.stl" in the "myterrain.zip" folder.** 

**I do still get the following error though...**

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-12-d2e750ee24f6> in <module>
      4 # This may take some time! You'll see In[*] and some log messages (will also be in the logfile inside the zip)
      5 # You may see some red stuff with 10%, etc. - don't worry, that's normal
----> 6 totalsize, full_zip_file_name = TouchTerrain.get_zipped_tiles(**args) # all args are in a dict
      7 print("\nDONE!\n\nCreated zip file", full_zip_file_name,  "%.2f" % totalsize, "Mb")
      8 

TypeError: cannot unpack non-iterable NoneType object

**Also, (maybe I should make this a different issue) when I switch to the Google Earth Engine's part of TouchTerrain (i.e. using part B)....**
```
    # A: use local DEM raster (geotiff)
    #"importedDEM": "pyramid.tif",  # put file in same folder as this notebook file!
    
    # B: use area and a DEM online source via EarthEngine
     "importedDEM": None,
```
**I get the following error...**

---------------------------------------------------------------------------
UnboundLocalError                         Traceback (most recent call last)
<ipython-input-21-d2e750ee24f6> in <module>
      4 # This may take some time! You'll see In[*] and some log messages (will also be in the logfile inside the zip)
      5 # You may see some red stuff with 10%, etc. - don't worry, that's normal
----> 6 totalsize, full_zip_file_name = TouchTerrain.get_zipped_tiles(**args) # all args are in a dict
      7 print("\nDONE!\n\nCreated zip file", full_zip_file_name,  "%.2f" % totalsize, "Mb")
      8 

~\Downloads\TouchTerrain_for_CAGEO-master\common\TouchTerrainEarthEngine.py in get_zipped_tiles(DEM_name, trlat, trlon, bllat, bllon, importedDEM, printres, ntilesx, ntilesy, tilewidth, basethick, zscale, fileformat, tile_centered, CPU_cores_to_use, max_cells_for_memory_only, temp_folder, zip_file_name, no_bottom, bottom_image, ignore_leq, unprojected, only, original_query_string, no_normals, projection, **otherargs)
    333 
    334         # get center of region as lon/lat, needed for conversion to meters
--> 335         center = [(region[0][0] + region[1][0])/2.0, (region[0][1] + region[2][1])/2.0]
    336 
    337         # Make a more descriptive name for the selected DEM from it's official (ee) name and the center

UnboundLocalError: local variable 'region' referenced before assignment

**Any thoughts?**

ChHarding, 2020-01-03:
Good work, I guess we’re nearly there for the local raster processing part! 

I assume the zip file/folder you get also contains the logfile.txt - what are the last couple of lines in that?

As to the unpack type error, I’m guessing that one or both of the two returns from get_zipped_tiles() are None

As you’re using the jupiter notebook version (right?), you can’t set a debugger breakpoint at the end of get_zipped_tiles so just put in this before the return:

print(“total size is”, total_size)
print(“zip file name is”, full_zip_file_name)

return total_size, full_zip_file_name

Regarding your issue with B:

what args are you using? Just the stuff that’s in the example i.e.:

# B: use area and a DEM online source via EarthEngine
    "importedDEM": None,
    "DEM_name": "USGS/NED",   # DEM source
    "bllat": 44.50185267072875,   # bottom left corner lat
    "bllon": -108.25427910156247, # bottom left corner long
    "trlat": 44.69741706507476,   # top right corner lat
    "trlon": -107.97962089843747, # top right corner long
	etc.

The local variable 'region' referenced before assignment error in 339 means that region hasn’t been defined which it should have done in 333  

You could print out bllon, bllat, trlat, trlon before 333 and see if they exist (the should be arguments to get_zipped_tiles …)

Also, is the import ee  (for the Earth Engine module) working or do you by any change get the Earth Engine module (ee) not installed error?

Did you get a dev license from Google to run Earth Engine?

> 
> That worked for me! With that I end up with "pyramid_tile_1_1.stl" in the "myterrain.zip" folder.
> 
> I do still get the following error though...
> 
> TypeError Traceback (most recent call last)
> in 
> 4 # This may take some time! You'll see In[*] and some log messages (will also be in the logfile inside the zip)
> 5 # You may see some red stuff with 10%, etc. - don't worry, that's normal
> ----> 6 totalsize, full_zip_file_name = TouchTerrain.get_zipped_tiles(**args) # all args are in a dict
> 7 print("\nDONE!\n\nCreated zip file", full_zip_file_name, "%.2f" % totalsize, "Mb")
> 8
> 
> TypeError: cannot unpack non-iterable NoneType object

> Also, (maybe I should make this a different issue) when I switch to the Google Earth Engine's part of TouchTerrain (i.e. using part B)....
> 
>     # A: use local DEM raster (geotiff)
>     #"importedDEM": "pyramid.tif",  # put file in same folder as this notebook file!
>     
>     # B: use area and a DEM online source via EarthEngine
>      "importedDEM": None,
> I get the following error...
> 
> UnboundLocalError Traceback (most recent call last)
> in 
> 4 # This may take some time! You'll see In[*] and some log messages (will also be in the logfile inside the zip)
> 5 # You may see some red stuff with 10%, etc. - don't worry, that's normal
> ----> 6 totalsize, full_zip_file_name = TouchTerrain.get_zipped_tiles(**args) # all args are in a dict
> 7 print("\nDONE!\n\nCreated zip file", full_zip_file_name, "%.2f" % totalsize, "Mb")
> 8
> 
> ~\Downloads\TouchTerrain_for_CAGEO-master\common\TouchTerrainEarthEngine.py in get_zipped_tiles(DEM_name, trlat, trlon, bllat, bllon, importedDEM, printres, ntilesx, ntilesy, tilewidth, basethick, zscale, fileformat, tile_centered, CPU_cores_to_use, max_cells_for_memory_only, temp_folder, zip_file_name, no_bottom, bottom_image, ignore_leq, unprojected, only, original_query_string, no_normals, projection, **otherargs)
> 333
> 334 # get center of region as lon/lat, needed for conversion to meters
> --> 335 center = [(region[0][0] + region[1][0])/2.0, (region[0][1] + region[2][1])/2.0]
> 336
> 337 # Make a more descriptive name for the selected DEM from it's official (ee) name and the center
> 
> UnboundLocalError: local variable 'region' referenced before assignment
> 
> Any thoughts?

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

ljverge, 2020-01-04:
_I assume the zip file/folder you get also contains the logfile.txt - what are the last couple of lines in that?_

**1 x 1 tiles, tile size 80.00 x 80.00 mm**
 
**tile 1 1 : height:  0.5 - 1795.4656 mm , file size: 8 Mb** 

**total size for all tiles: 8 Mb** 

**processing finished: 11:47:50.283236** 
 
 

_print(“total size is”, total_size)
print(“zip file name is”, full_zip_file_name)

return total_size, full_zip_file_name_

 **I could never get this to work, but it didn't affect the end result.** 
 
 

_What args are you using? Just the stuff that’s in the example i.e.:_

**Yes, I was just trying to get the example data to work.** 
 
 
**I did just finally managed to get everything besides the _cannot unpack non-iterable NoneType object_ error to work. I'm not sure all the little modifications that I made along the way (I really should have kept track), but I was able to create an STL from my selected coordinates. Hey, that's the goal right :)**

**Thanks for all your help!**

ChHarding, 2020-01-06:
So what do the print()s you put in tell you? Is one of them None? I’m really curious :) Please try this

print(“DEBUG: total size is”, total_size)
print(“DEBUGL zip file name is”, full_zip_file_name)

return total_size, full_zip_file_name

 and see what they print.

> I assume the zip file/folder you get also contains the logfile.txt - what are the last couple of lines in that?
>
> 1 x 1 tiles, tile size 80.00 x 80.00 mm
>
> tile 1 1 : height: 0.5 - 1795.4656 mm , file [...]

What was the issue with region?

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

ChHarding, 2020-01-14:
Closing this issue. Will issue a patch with the fix for the writeable problem. Not sure what the details for last 2 issue were (I don't have them) so I'm closing this for now.

closed 2020-01-14 by ChHarding

ljverge, 2020-01-15:
**Sorry for my delayed response. I was unsure where exactly you wanted me to add in the "print" commands. When I enter them here...**

```
from common import TouchTerrainEarthEngine as TouchTerrain
# Process the data

# This may take some time! You'll see In[*] and some log messages (will also be in the logfile inside the zip)
# You may see some red stuff with 10%, etc. - don't worry, that's normal
totalsize, full_zip_file_name = TouchTerrain.get_zipped_tiles(**args) # all args are in a dict
print(“DEBUG: total size is”, total_size)
print(“DEBUGL zip file name is”, full_zip_file_name)

return total_size, full_zip_file_name
print("\nDONE!\n\nCreated zip file", full_zip_file_name,  "%.2f" % totalsize, "Mb")

# your zip file will be inside the tmp folder which is inside the same folder your notebook file is in
```

**I get this error...**

  _File "<ipython-input-6-1932bfd53dde>", line 7
    print(“DEBUG: total size is”, total_size)
               ^
SyntaxError: invalid character in identifier_

ChHarding, 2020-01-16:
Sorry about that! Remove the 2 lines you added to the notebook and instead add them in at the end of the get_zipped_tiles() function in TouchTerrainEarthEngine.py,  just before it returns those 2 variables.For me that’s about line 1111. 

Change

return total_size, full_zip_file_name

to 

print(“DEBUG: total size is”, total_size)
print(“DEBUG: zip file name is”, full_zip_file_name)
return total_size, full_zip_file_name

run that and see what it prints out

> 
> Sorry for my delayed response. I was unsure where exactly you wanted me to add in the "print" commands. When I enter them here...
> 
> from common import TouchTerrainEarthEngine as TouchTerrain
> # Process the data
> 
> # This may take some time! You'll see In[*] and some log messages (will also be in the logfile inside the zip)
> # You may see some red stuff with 10%, etc. - don't worry, that's normal
> totalsize, full_zip_file_name = TouchTerrain.get_zipped_tiles(**args) # all args are in a dict
> print(“DEBUG: total size is”, total_size)
> print(“DEBUGL zip file name is”, full_zip_file_name)
> 
> return total_size, full_zip_file_name
> print("\nDONE!\n\nCreated zip file", full_zip_file_name,  "%.2f" % totalsize, "Mb")
> 
> # your zip file will be inside the tmp folder which is inside the same folder your notebook file is in
> I get this error...
> 
> File "", line 7
> print(“DEBUG: total size is”, total_size)
> ^
> SyntaxError: invalid character in identifier

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

ljverge, 2020-01-16:
**I get all kinds of crazy errors when I try that. I'm not sure if it's a syntax thing.** 

> ---------------------------------------------------------------------------
> SyntaxError                               Traceback (most recent call last)
> C:\Python\lib\site-packages\IPython\core\interactiveshell.py in run_code(self, code_obj, result, async_)
>    3325                 else:
> -> 3326                     exec(code_obj, self.user_global_ns, self.user_ns)
>    3327             finally:
> 
> <ipython-input-7-d2e750ee24f6> in <module>
> ----> 1 from common import TouchTerrainEarthEngine as TouchTerrain
>       2 # Process the data
>       3 
> 
> SyntaxError: invalid character in identifier (TouchTerrainEarthEngine.py, line 1113)
> 
> During handling of the above exception, another exception occurred:
> 
> UnicodeDecodeError                        Traceback (most recent call last)
> C:\Python\lib\site-packages\IPython\core\interactiveshell.py in run_code(self, code_obj, result, async_)
>    3341             if result is not None:
>    3342                 result.error_in_exec = sys.exc_info()[1]
> -> 3343             self.showtraceback(running_compiled_code=True)
>    3344         else:
>    3345             outflag = False
> 
> C:\Python\lib\site-packages\IPython\core\interactiveshell.py in showtraceback(self, exc_tuple, filename, tb_offset, exception_only, running_compiled_code)
>    2024                 # Though this won't be called by syntax errors in the input
>    2025                 # line, there may be SyntaxError cases with imported code.
> -> 2026                 self.showsyntaxerror(filename, running_compiled_code)
>    2027             elif etype is UsageError:
>    2028                 self.show_usage_error(value)
> 
> C:\Python\lib\site-packages\IPython\core\interactiveshell.py in showsyntaxerror(self, filename, running_compiled_code)
>    2086         # If the error occurred when executing compiled code, we should provide full stacktrace.
>    2087         elist = traceback.extract_tb(last_traceback) if running_compiled_code else []
> -> 2088         stb = self.SyntaxTB.structured_traceback(etype, value, elist)
>    2089         self._showtraceback(etype, value, stb)
>    2090 
> 
> C:\Python\lib\site-packages\IPython\core\ultratb.py in structured_traceback(self, etype, value, elist, tb_offset, context)
>    1418                 and isinstance(value.lineno, int):
>    1419             linecache.checkcache(value.filename)
> -> 1420             newtext = linecache.getline(value.filename, value.lineno)
>    1421             if newtext:
>    1422                 value.text = newtext
> 
> C:\Python\lib\linecache.py in getline(filename, lineno, module_globals)
>      14 
>      15 def getline(filename, lineno, module_globals=None):
> ---> 16     lines = getlines(filename, module_globals)
>      17     if 1 <= lineno <= len(lines):
>      18         return lines[lineno-1]
> 
> C:\Python\lib\linecache.py in getlines(filename, module_globals)
>      45 
>      46     try:
> ---> 47         return updatecache(filename, module_globals)
>      48     except MemoryError:
>      49         clearcache()
> 
> C:\Python\lib\linecache.py in updatecache(filename, module_globals)
>     135     try:
>     136         with tokenize.open(fullname) as fp:
> --> 137             lines = fp.readlines()
>     138     except OSError:
>     139         return []
> 
> C:\Python\lib\codecs.py in decode(self, input, final)
>     320         # decode input (taking the buffer into account)
>     321         data = self.buffer + input
> --> 322         (result, consumed) = self._buffer_decode(data, self.errors, final)
>     323         # keep undecoded input until the next call
>     324         self.buffer = data[consumed:]
> 
> UnicodeDecodeError: 'utf-8' codec can't decode byte 0x93 in position 4918: invalid start byte

**Here's the code that I added**
```
                    try:    
                    	os.remove(log_file_name) 
                    except Exception as e:   
         #print("Error removing logfile " + str(log_file_name) + " " + str(e), file=sys.stderr)     
             
    # return total  size in Mega bytes and location of zip file
                    	print(“DEBUG: total size is”, total_size)
                    	print(“DEBUG: zip file name is”, full_zip_file_name)
                    	return total_size, full_zip_file_name
#
# MAIN
#
if __name__ == "__main__":
```

ChHarding, 2020-01-16:
You have to make sure that the indentations match with the original return. Don’t use tabs, instead use spaces 

Instead of
    # return total  size in Mega bytes and location of zip file
                    	print(“DEBUG: total size is”, total_size)
                    	print(“DEBUG: zip file name is”, full_zip_file_name)
                    	return total_size, full_zip_file_name
try:
    # return total  size in Mega bytes and location of zip file
    print(“DEBUG: total size is”, total_size)
    print(“DEBUG: zip file name is”, full_zip_file_name)
    return total_size, full_zip_file_name

> 
> I get all kinds of crazy errors when I try that. I'm not sure if it's a syntax thing.
> 
> SyntaxError Traceback (most recent call last)
> C:\Python\lib\site-packages\IPython\core\interactiveshell.py in run_code(self, code_obj, result, async_)
> 3325 else:
> -> 3326 exec(code_obj, self.user_global_ns, self.user_ns)
> 3327 finally:
> 
> in 
> ----> 1 from common import TouchTerrainEarthEngine as TouchTerrain
> 2 # Process the data
> 3
> 
> SyntaxError: invalid character in identifier (TouchTerrainEarthEngine.py, line 1113)
> 
> During handling of the above exception, another exception occurred:
> 
> UnicodeDecodeError Traceback (most recent call last)
> C:\Python\lib\site-packages\IPython\core\interactiveshell.py in run_code(self, code_obj, result, async_)
> 3341 if result is not None:
> 3342 result.error_in_exec = sys.exc_info()[1]
> -> 3343 self.showtraceback(running_compiled_code=True)
> 3344 else:
> 3345 outflag = False
> 
> C:\Python\lib\site-packages\IPython\core\interactiveshell.py in showtraceback(self, exc_tuple, filename, tb_offset, exception_only, running_compiled_code)
> 2024 # Though this won't be called by syntax errors in the input
> 2025 # line, there may be SyntaxError cases with imported code.
> -> 2026 self.showsyntaxerror(filename, running_compiled_code)
> 2027 elif etype is UsageError:
> 2028 self.show_usage_error(value)
> 
> C:\Python\lib\site-packages\IPython\core\interactiveshell.py in showsyntaxerror(self, filename, running_compiled_code)
> 2086 # If the error occurred when executing compiled code, we should provide full stacktrace.
> 2087 elist = traceback.extract_tb(last_traceback) if running_compiled_code else []
> -> 2088 stb = self.SyntaxTB.structured_traceback(etype, value, elist)
> 2089 self._showtraceback(etype, value, stb)
> 2090
> 
> C:\Python\lib\site-packages\IPython\core\ultratb.py in structured_traceback(self, etype, value, elist, tb_offset, context)
> 1418 and isinstance(value.lineno, int):
> 1419 linecache.checkcache(value.filename)
> -> 1420 newtext = linecache.getline(value.filename, value.lineno)
> 1421 if newtext:
> 1422 value.text = newtext
> 
> C:\Python\lib\linecache.py in getline(filename, lineno, module_globals)
> 14
> 15 def getline(filename, lineno, module_globals=None):
> ---> 16 lines = getlines(filename, module_globals)
> 17 if 1 <= lineno <= len(lines):
> 18 return lines[lineno-1]
> 
> C:\Python\lib\linecache.py in getlines(filename, module_globals)
> 45
> 46 try:
> ---> 47 return updatecache(filename, module_globals)
> 48 except MemoryError:
> 49 clearcache()
> 
> C:\Python\lib\linecache.py in updatecache(filename, module_globals)
> 135 try:
> 136 with tokenize.open(fullname) as fp:
> --> 137 lines = fp.readlines()
> 138 except OSError:
> 139 return []
> 
> C:\Python\lib\codecs.py in decode(self, input, final)
> 320 # decode input (taking the buffer into account)
> 321 data = self.buffer + input
> --> 322 (result, consumed) = self._buffer_decode(data, self.errors, final)
> 323 # keep undecoded input until the next call
> 324 self.buffer = data[consumed:]
> 
> UnicodeDecodeError: 'utf-8' codec can't decode byte 0x93 in position 4918: invalid start byte
> 
> Here's the code that I added
> 
>                     try:    
>                     	os.remove(log_file_name) 
>                     except Exception as e:   
>          #print("Error removing logfile " + str(log_file_name) + " " + str(e), file=sys.stderr)     
>              
>     # return total  size in Mega bytes and location of zip file
>                     	print(“DEBUG: total size is”, total_size)
>                     	print(“DEBUG: zip file name is”, full_zip_file_name)
>                     	return total_size, full_zip_file_name
> #
> # MAIN
> #
> if __name__ == "__main__":

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

ljverge, 2020-01-16:
**Sorry, same error.**

> SyntaxError: invalid character in identifier (TouchTerrainEarthEngine.py, line 1113)

ChHarding, 2020-01-16:
Can you send me your TouchTerrainEarthEngine.py ?

> 
> Sorry, same error.
> 
> SyntaxError: invalid character in identifier (TouchTerrainEarthEngine.py, line 1113)

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu