ChHarding/TouchTerrain_for_CAGEO #25: Website error

karsunky, 2020-02-19:
When using website with any args the following error is thrown:

Error:,Projection: Argument 'crs': Invalid type. Expected: String. Actual: Type.

Tried:
Changing source
Adding param "projection":26913
"projection":null
All tests generate same err

ChHarding, 2020-02-19:
Thanks, we are looking at it now. Not sure why it suddenly doesn’t like the projection string but I think it’s something else. Sorry about the mess! I’ll email you when we’ve fixed it.  

> 
> When using website with any args the following error is thrown:
> 
> Error:,Projection: Argument 'crs': Invalid type. Expected: String. Actual: Type.
> 
> Tried:
> Changing source
> Adding param "projection":26913
> "projection":null
> All tests generate same err

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

ChHarding, 2020-02-19:
I fixed the problem with this issue, however, there’s still the issue of not getting the hill shaded overlay in the Google Maps window. Sorry about the outage - you should be able to use the app again, just be aware that you won’t see the grey-is overlay in the map window for now.

BTW,  just for my interest: What are you using TouchTerrain for?

> 
> When using website with any args the following error is thrown:
> 
> Error:,Projection: Argument 'crs': Invalid type. Expected: String. Actual: Type.
> 
> Tried:
> Changing source
> Adding param "projection":26913
> "projection":null
> All tests generate same err

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

karsunky, 2020-02-19:
Thanks so much!
Oddly enough I was just trying to do an export.

Jon

ChHarding, 2020-02-19:
Well, what I meant is what do you use the exported STLs for? Teaching, research, hobby …. ?

> 
> Thanks so much!
> Oddly enough I was just trying to do an export.
> 
> 
> Jon
> 
> 
> > I fixed the problem with this issue, however, there’s still the issue of
> > not getting the hill shaded overlay in the Google Maps window. Sorry about
> > the outage - you should be able to use the app again, just be aware that
> > you won’t see the grey-is overlay in the map window for now.
> >
> >
> > BTW, just for my interest: What are you using TouchTerrain for?
> >
> > >
> > > When using website with any args the following error is thrown:
> > >
> > > Error:,Projection: Argument 'crs': Invalid type. Expected: String.
> > Actual: Type.
> > >
> > > Tried:
> > > Changing source
> > > Adding param "projection":26913
> > > "projection":null
> > > All tests generate same err
> > >
> > https://github.com/ChHarding/TouchTerrain_for_CAGEO/issues/25?email_source=notifications&email_token=AEYDF5JOXP7PF47TT7LS5ATRDV7RBA5CNFSM4KX7ZLAKYY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4IOXUU5A <https://github.com/ChHarding/TouchTerrain_for_CAGEO/issues/25?email_source=notifications&email_token=AEYDF5JOXP7PF47TT7LS5ATRDV7RBA5CNFSM4KX7ZLAKYY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4IOXUU5A>>,
> > >
> >
> > Chris Harding
> > Associate Professor
> > Department of Geological & Atmospheric Sciences
> >
> > TouchTerrain.geol.iastate.edu <http://touchterrain.geol.iastate.edu/>
> >
> >
> > <https://github.com/ChHarding/TouchTerrain_for_CAGEO/issues/25?email_source=notifications&email_token=ADTNY7VMGT6N4GWHNMKTW53RDW5UXA5CNFSM4KX7ZLAKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEMKDK5A#issuecomment-588526964 <https://github.com/ChHarding/TouchTerrain_for_CAGEO/issues/25?email_source=notifications&email_token=ADTNY7VMGT6N4GWHNMKTW53RDW5UXA5CNFSM4KX7ZLAKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEMKDK5A#issuecomment-588526964>>,

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

karsunky, 2020-02-20:
Thanks Chris , when I get to the train animation it spins indefinitely.
I'll try a few different combinations and see if I can nail it down

*DEM_name = USGS/NEDtrlat = 46.23893176436362trlon =
-122.12780309616276bllat = 46.154093072931545bllon =
-122.25174261032292printres = 0.35ntilesx = 1ntilesy = 1tilewidth =
300.0basethick = 1.0zscale = 1.25fileformat = STLbno_bottom = True *

I just read your bio, your work looks interesting. I used to be on the MS
Hololens Team and have one on the shelf I need to dust off. I'd imagine
this tool could be very useful in creating AR assets

*Send me your mailing address if you'd like a 3D gift *

[image: SnoqValleyWA 0 20200118 00xx.jpg]

Jon Karsunky
Seattle, WA

karsunky, 2020-02-20:
Test results:
Default website parameters work and export button is shown

Changing the printres and tilewidth cause the website to process/hang
indefinitely. Export is never shown

*DEFAULT params*
DEM_name = USGS/NED
trlat = 44.697417065074774
trlon = -107.9796208984375
bllat = 44.50185267072878
bllon = -108.2542791015625
printres = 0.4
ntilesx = 1
ntilesy = 1
tilewidth = 80.0
basethick = 1.0
zscale = 1.0
fileformat = STLb

*Changing printres and tilewidth causes hang*
DEM_name = USGS/NED
trlat = 44.697417065074774
trlon = -107.9796208984375
bllat = 44.50185267072878
bllon = -108.2542791015625
*printres = 0.4*
ntilesx = 1
ntilesy = 1
*tilewidth = 300.0*
basethick = 1.0
zscale = 1.0
fileformat = STLb

On Wed, Feb 19, 2020 at 4:05 PM K, Jon <jonk@aspdeveloper.com> wrote:

> Thanks Chris , when I get to the train animation it spins indefinitely.
> I'll try a few different combinations and see if I can nail it down
>
>
>
>
>
>
>
>
>
>
>
>
>
> *DEM_name = USGS/NEDtrlat = [...]

ChHarding, 2020-02-20:
You’re right, with larger models if hangs, I’ll look into this tomorrow ...

> 
> Test results:
> Default website parameters work and export button is shown
> 
> Changing the printres and tilewidth cause the website to process/hang
> indefinitely. Export is never shown
> 
> *DEFAULT params*
> DEM_name = USGS/NED
> trlat = 44.697417065074774
> trlon = -107.9796208984375
> bllat = 44.50185267072878
> bllon = -108.2542791015625
> printres = 0.4
> ntilesx = 1
> ntilesy = 1
> tilewidth = 80.0
> basethick = 1.0
> zscale = 1.0
> fileformat = STLb
> 
> 
> *Changing printres and tilewidth causes hang*
> DEM_name = USGS/NED
> trlat = 44.697417065074774
> trlon = -107.9796208984375
> bllat = 44.50185267072878
> bllon = -108.2542791015625
> *printres = 0.4*
> ntilesx = 1
> ntilesy = 1
> *tilewidth = 300.0*
> basethick = 1.0
> zscale = 1.0
> fileformat = STLb
> 
> 
> 
> >
> > Thanks Chris , when I get to the train animation it spins indefinitely.
> > I'll try a few different combinations and see if I can nail it down
> >
> >
> >
> >
> >
> >
> >
> >
> >
> >
> >
> >
> >
> > *DEM_name = USGS/NEDtrlat = 46.23893176436362trlon =
> > -122.12780309616276bllat = 46.154093072931545bllon =
> > -122.25174261032292printres = 0.35ntilesx = 1ntilesy = 1tilewidth =
> > 300.0basethick = 1.0zscale = 1.25fileformat = STLbno_bottom = True *
> >
> >
> > I just read your bio, your work looks interesting. I used to be on the MS
> > Hololens Team and have one on the shelf I need to dust off. I'd imagine
> > this tool could be very useful in creating AR assets
> >
> > *Send me your mailing address if you'd like a 3D gift *
> >
> > [image: SnoqValleyWA 0 20200118 00xx.jpg]
> >
> > Jon Karsunky
> > Seattle, WA

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

ChHarding, 2020-02-20:
Try it know, it looks like we fixed it.  But we’ve not tested if well with several users AND large models so it may still hang for you if you’re unlucky. But let me know if that happens a lot, OK?

> 
> Test results:
> Default website parameters work and export button is shown
> 
> Changing the printres and tilewidth cause the website to process/hang
> indefinitely. Export is never shown
> 
> *DEFAULT params*
> DEM_name = USGS/NED
> trlat = 44.697417065074774
> trlon = -107.9796208984375
> bllat = 44.50185267072878
> bllon = -108.2542791015625
> printres = 0.4
> ntilesx = 1
> ntilesy = 1
> tilewidth = 80.0
> basethick = 1.0
> zscale = 1.0
> fileformat = STLb
> 
> 
> *Changing printres and tilewidth causes hang*
> DEM_name = USGS/NED
> trlat = 44.697417065074774
> trlon = -107.9796208984375
> bllat = 44.50185267072878
> bllon = -108.2542791015625
> *printres = 0.4*
> ntilesx = 1
> ntilesy = 1
> *tilewidth = 300.0*
> basethick = 1.0
> zscale = 1.0
> fileformat = STLb
> 
> 
> 
> >
> > Thanks Chris , when I get to the train animation it spins indefinitely.
> > I'll try a few different combinations and see if I can nail it down
> >
> >
> >
> >
> >
> >
> >
> >
> >
> >
> >
> >
> >
> > *DEM_name = USGS/NEDtrlat = 46.23893176436362trlon =
> > -122.12780309616276bllat = 46.154093072931545bllon =
> > -122.25174261032292printres = 0.35ntilesx = 1ntilesy = 1tilewidth =
> > 300.0basethick = 1.0zscale = 1.25fileformat = STLbno_bottom = True *
> >
> >
> > I just read your bio, your work looks interesting. I used to be on the MS
> > Hololens Team and have one on the shelf I need to dust off. I'd imagine
> > this tool could be very useful in creating AR assets
> >
> > *Send me your mailing address if you'd like a 3D gift *
> >
> > [image: SnoqValleyWA 0 20200118 00xx.jpg]
> >
> > Jon Karsunky
> > Seattle, WA

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

karsunky, 2020-02-21:
Working great, thanks!

ChHarding, 2020-02-21:
Glad to hear it, thanks!

As to our earlier chat:

- My university web page is horribly outdated, I used to do geoscience VR but now I pretty much only do 3D terrain stuff for teaching purposes.  

- Hence my question about how you use the TouchTerrain models. It’s totally OK if you just make them b/c they are interesting (or to sell them ;) but I’m genuinely curious to find out how people use them. I’ve gathered a few use cases already but as you apparently have a (former?) tech background I’d be especially interested how you use them. E.g. you seem to use WA terrain which makes sense as users often report printing areas “around” them …

- If you want send me a print, my address is:

Chris Harding
Iowa State University
1620 Howe Hall
Ames, IA 50011

Cheers

Chris

> 
> Working great, thanks!
> 
> 
> > Try it now it looks like we fixed it. But we’ve not tested if well with
> > several users AND large models so it may still hang for you if you’re
> > unlucky. But let me know if that happens a lot, OK?
> >
> > >
> > > Test results:
> > > Default website parameters work and export button is shown
> > >
> > > Changing the printres and tilewidth cause the website to process/hang
> > > indefinitely. Export is never shown
> > >
> > > *DEFAULT params*
> > > DEM_name = USGS/NED
> > > trlat = 44.697417065074774
> > > trlon = -107.9796208984375
> > > bllat = 44.50185267072878
> > > bllon = -108.2542791015625
> > > printres = 0.4
> > > ntilesx = 1
> > > ntilesy = 1
> > > tilewidth = 80.0
> > > basethick = 1.0
> > > zscale = 1.0
> > > fileformat = STLb
> > >
> > >
> > > *Changing printres and tilewidth causes hang*
> > > DEM_name = USGS/NED
> > > trlat = 44.697417065074774
> > > trlon = -107.9796208984375
> > > bllat = 44.50185267072878
> > > bllon = -108.2542791015625
> > > *printres = 0.4*
> > > ntilesx = 1
> > > ntilesy = 1
> > > *tilewidth = 300.0*
> > > basethick = 1.0
> > > zscale = 1.0
> > > fileformat = STLb
> > >
> > >
> > >
> > > >
> > > > Thanks Chris , when I get to the train animation it spins indefinitely.
> > > > I'll try a few different combinations and see if I can nail it down
> > > >
> > > >
> > > >
> > > >
> > > >
> > > >
> > > >
> > > >
> > > >
> > > >
> > > >
> > > >
> > > >
> > > > *DEM_name = USGS/NEDtrlat = 46.23893176436362trlon =
> > > > -122.12780309616276bllat = 46.154093072931545bllon =
> > > > -122.25174261032292printres = 0.35ntilesx = 1ntilesy = 1tilewidth =
> > > > 300.0basethick = 1.0zscale = 1.25fileformat = STLbno_bottom = True *
> > > >
> > > >
> > > > I just read your bio, your work looks interesting. I used to be on the
> > MS
> > > > Hololens Team and have one on the shelf I need to dust off. I'd imagine
> > > > this tool could be very useful in creating AR assets
> > > >
> > > > *Send me your mailing address if you'd like a 3D gift *
> > > >
> > > > [image: SnoqValleyWA 0 20200118 00xx.jpg]
> > > >
> > > > Jon Karsunky
> > > > Seattle, WA
> > > >
> > https://github.com/ChHarding/TouchTerrain_for_CAGEO/issues/25?email_source=notifications&email_token=AEYDF5MENGZPFGBEW2GAGDTRDXDJ3A5CNFSM4KX7ZLAKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEMKG6QI#issuecomment-588541761>,
> > >
> >
> > Chris Harding
> > Associate Professor
> > Department of Geological & Atmospheric Sciences
> >
> > TouchTerrain.geol.iastate.edu

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

closed 2021-05-05 by ChHarding