ChHarding/TouchTerrain_for_CAGEO #26: Error exporting to STLb: ,Invalid value at 'thumbnail.file_format' (TYPE_ENUM), "TIFF"

mpcrossley, 2020-03-08:
I've tried a multitude of export options but can't seem to generate anything. Is there something I'm doing wrong?

[image]

ChHarding, 2020-03-09:
No,  I don’t think that it’s anything you’re doing  wrong.  Something in the underlying web structure changed recently. Thanks for reporting, we’re working on it. I’ll let you know once we’ve fixed it.

> 
> I've tried a multitude of export options but can't seem to generate anything. Is there something I'm doing wrong?
> 
>  <https://user-images.githubusercontent.com/9376945/76157398-428f6280-60bd-11ea-8df3-de8cf7495519.png>

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

ChHarding, 2020-03-11:
Should work again now, sorry for the outage!  

Just curious - can I ask you what you use the terrain models for?

> 
> I've tried a multitude of export options but can't seem to generate anything. Is there something I'm doing wrong?
> 
>  <https://user-images.githubusercontent.com/9376945/76157398-428f6280-60bd-11ea-8df3-de8cf7495519.png>

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

mpcrossley, 2020-03-13:
Sure!

My parents did a lot of volunteering in Kenya (Kisumu region), and I was going to have myself and their grandkids print out a topo 3d print of the region and paint it for their anniversary.

ChHarding, 2020-03-13:
Sounds great! Don’t forget to prime it before painting,  I use simple grey automotive primer filler spay.

> 
> 
> Sure!
> 
> My parents did a lot of volunteering in Kenya (Kisumu region), and I was going to have myself and their grandkids print out a topo 3d print of the region and paint it for their anniversary.
> From: ChHarding <notifications@github.com>
> Sent: Wednesday, March 11, 2020 9:37:53 AM
> To: ChHarding/TouchTerrain_for_CAGEO <TouchTerrain_for_CAGEO@noreply.github.com>
> Cc: Matt Crossley <matt@mattcrossley.com>; Author <author@noreply.github.com>
> 
> 
> Should work again now, sorry for the outage!
> 
> Just curious - can I ask you what you use the terrain models for?
> 
> >
> > I've tried a multitude of export options but can't seem to generate anything. Is there something I'm doing wrong?
> >
> > <https://user-images.githubusercontent.com/9376945/76157398-428f6280-60bd-11ea-8df3-de8cf7495519.png>
> >
> 
> Chris Harding
> Associate Professor
> Department of Geological & Atmospheric Sciences
> 
> TouchTerrain.geol.iastate.edu

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

mpcrossley, 2020-03-13:
Awesome thanks for the hint!

closed 2021-05-05 by ChHarding