ChHarding/TouchTerrain_for_CAGEO #27: Allow lowering of water

idenc, 2020-04-21:
Add a configuration option to lower the height of oceans for emphasis? I wouldn't mind helping to implement this if someone could point me in the right direction

ChHarding, 2020-04-21:
What do you mean with lowering?  I have an option ignore_leq to remove all cells with elevation values less that a certain value. so if you set this to 0 (or maybe 0.5) (“ignore_leq”:0) in Manual settings) and request an area with a coastline, the model won’t contain any water:

Or are you thinking of still keeping them but lowering them to say -50? That would no be difficult to add.

Cheers

Chris

> 
> 
> Add a configuration option to lower the height of oceans for emphasis? I wouldn't mind helping to implement this if someone could point me in the right direction

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

idenc, 2020-04-21:
I'd like to keep the water, but lower them by some amount so that the coastline, for example, wouldn't blend with the water so much. Here is an example of a model generated with other software that has this option, vs. generating it without
[image]
[image]

ChHarding, 2020-04-21:
Oh, interesting! What’s that software?

I can probably do this with another manual option something like:

 “lower_leq”:[0, <offset>]

where offset would be the value of lowering any cells that are <= 0 (in this case)

Question is what the unit should be? Probably in meters?

Also, z-scaling would still apply to this offset, right? So if is given as 10 , I would still multiply it by the z-scale factor and then lower it by that value ….

> I'd like to keep the water, but lower them by some amount so that the coastline, for example, wouldn't blend with the water so much. Here is an example of a model generated with other software that [...]

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

idenc, 2020-04-21:
That software was [Terrain2STL](http://jthatch.com/Terrain2STL/). They use a drop of mm, based on the physical size I assume. I suppose it doesn't matter if z-scaling applies since the offset could be adjusted accordingly, it just needs to be clear whether it applies or not.

ChHarding, 2020-04-21:
So you think using mm is better?

> 
> 
> That software was Terrain2STL <http://jthatch.com/Terrain2STL/>. They use a drop of mm, based on the physical size I assume. I suppose it doesn't matter if z-scaling applies since the offset could be adjusted accordingly, it just needs to be clear whether it applies or not.

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

idenc, 2020-04-22:
Yes, I think so. It's easier for me to visualize how the final model will turn out when printed at least

ChHarding, 2020-05-11:
Sorry for the long silence!  I looked at your changes and did merge them by hand b/c there were a lot of other chances I already worked on. Your code works fine except it did not account for a z-scale != 1.0, so the desired mm offset would have later been multiplied by this z-scale.

With this, your pull request is no longer needed. However, I would like to ask you if you could have a look at the current beta:  https://github.com/ChHarding/TouchTerrain_for_CAGEO/tree/beta

We have restructured the project to now build and install a touchterrain module and use that to import stuff from common and/or server. There's no standalone folder anymore, both of the stand alone files are now at the root level. Documentation for both has been updated to reflect these changes. We've also integrated the tiny single file vectors module directly into common. setup.py now also has numpy as a prereq, as per your suggestion. 

To install we recommend using pip install . (<- dot!) in the same folder as setup.py will build and install touchterrain and all it's prereqs, and then remove its working folders again (which is nice!)

I know you already have a working solution for your original request already but would you mind doing a clean install of the current beta and test it? If that works for you, I'm ready to put these changes into master and update the web app with it.

Thanks!

closed 2020-10-17 by ChHarding

OschauLars, 2023-06-05:
I would like to come back to this. I checked around the Southeast Arm of the Yellowstone Lake. "lower_leq":[0,x] works fine for the lake. It does not work for the Yellowstone River itself. Maybe the altitude of a river is same as the surrounding area? If so rivers in mountain areas might have various altitudes, so "lower_leq":[0,x] might work for some areas, only. 

What is my goal: I plan to use a CNC router to cut a relief into wood. I would like to lower any water and fill the area later with Resin. Therefore I need some extra depth here to allow the Resin to have some room.

It would be great to have any kind of water picked by the algorithm and lowered by a fixed number of millimeters.

Greetings from Germany!!

Lars

ChHarding, 2023-06-06:
Can you sent me something so I can reproduce and debug this?  (I forgot what your setup is …)

ChHarding, 2023-06-13:
Lars,

I just looked at the situation and you’re right,  lower_leq[0, x] has only one threshold and so if you set the lake-level elevation of the YS lake as that threshold, any parts of the YS river that eventually gets higher than the lake level (as the river elevation rises towards the mountains they come from) will not be altered, There’s really no good way to do this in TouchTerrain but it would not be too difficult for me to do in ArcGIS be defining a mask and lowering any pixel inside it. I would need the geotiff(s) you want modified and to what new elevation you want the lake and rivers to be dropped.

Cheers

Chris