ChHarding/TouchTerrain_for_CAGEO #29: UnboundLocalError when Processing test Pyramid

brend2b, 2020-04-27:
I'm having some problems getting access to Earth Engine on Google's end. I have a token and everything but I cannot sign in to the code editor even though they have sent me an email saying I had access.

While I work this out, I was trying to use the sample pyramid .tif file. I've uncommented out the "importedDEM" in Section A of the args and commented out "importedDEM" in section B. When I move to the next cell, I am given an error when I run it.

I'm running Anaconda 3, Python 3.8.2, Pillow 7.0.0, numpy 1.18.1, gdal 3.0.4, vectors 1.0.0, earthengine-api 0.1.219

I included the args and error I am receiving. What can I do to fix this?

[args.txt](https://github.com/ChHarding/TouchTerrain_for_CAGEO/files/4537294/args.txt)
[error.txt](https://github.com/ChHarding/TouchTerrain_for_CAGEO/files/4537295/error.txt)

ChHarding, 2020-04-27:
To generate a STL from a local geotiff you don’t need access to earth engine at all. It’s likely that you have full access to earth engine, it’s just not yet used here.  The error you got has to do with a change/improvement that I put in recently but apparently didn’t  test well enough. It seems like an easy fix but I will need to reproduce it first.

1) When did you download the project from GitHub?
2) Are you running the standalone version using touch terrain_standalone.py and the args in a json file or are you using the jupiter notebook version?

Cheers

Chris

> I'm having some problems getting access to Earth Engine on Google's end. I have a token and everything but I cannot sign in to the code editor even though they have sent me an email saying I had [...]

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

ChHarding, 2020-05-10:
I responded to you on this issue via non-github. It was a bug (thanks!) and I think I fixed it.  Please try the Beta and contact me  in case of further problems

closed 2020-05-10 by ChHarding