ChHarding/TouchTerrain_for_CAGEO #30: Python version inconsistency between readme and documentation.

TiesdeKok, 2020-07-03:
The Github readme mentions that Python 3.6 is required, but the documentation (e.g. https://docs.google.com/document/d/1bS-N7elMMWU44LctQpMPbbNSurftY3fRealTwOK-5ME/edit) explicitly asks to install Python 2.7.

ChHarding, 2020-07-03:
Ties,

sorry for the mistake and thanks for catching it! I’ve updated the google doc to install Python 3.

Cheers

Chris 

> 
> 
> The Github readme mentions that Python 3.6 is required, but the documentation (e.g. https://docs.google.com/document/d/1bS-N7elMMWU44LctQpMPbbNSurftY3fRealTwOK-5ME/edit <https://docs.google.com/document/d/1bS-N7elMMWU44LctQpMPbbNSurftY3fRealTwOK-5ME/edit>) explicitly asks to install Python 2.7.

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

TiesdeKok, 2020-07-03:
Thanks a lot!

closed 2020-07-03 by TiesdeKok