ChHarding/TouchTerrain_for_CAGEO #36: Set base hight in reference to Vertical datum

DL4APT, 2020-12-27:
if you want to manually create adjacent tiles they can't be set to a common base height. 
The best solution would be an option that allows the following: that the lowest point of the tile is not referenced to base height but base height plus the height that this point is above the Vertical datum.

My goal is to print many tiles to have more or less a wall with  a mountain range relief on it (lager than 1m²) and the maximum file size limit would compromise the quality drastically.

ChHarding, 2020-12-27:
Yes, I can see how defining a custom vertical datum from which to offset the base height as useful and I don’t think it would be too difficult to add this in the near future, as long as you’re OK with using it via the manual options (e.g. “vertical_offset”: 123.45).  However, I’m curious how you would know the value for all your offsets since there’s no way to get this information from the front end. Or are you just going to do all our models first, look at their elevation stats in the log, calculate the offset from that and re-run them?

But, maybe you have not tried the following which might already give you the result you’re looking for:  
- set up you tiles in a way you want them.
- if you need more that 8 tiles for this use ntilesx and ntilesy to set them manually (e.g. “ntilesx”:10, “ntilesy”:12)
- this will ofc be way to much to process in one setting, so use the only manual option to only run that one tile e.g. “only”:[1,1] 
- repeat this for all your tiles  “only”:[1,2], …,  “only”:[1,12], “only”:[2,1], …. which is granted a bit tedious but maybe less that running everything twice to get the offsets done properly. Plus, if you use the tiles, they are guaranteed to fit together perfectly, whereas in your way you’ll have to generate the tile boundary coordinates externally and type them in to be precise. (OK, maybe you don’t care about super fine precision …)

Or did you try this and it didn’t work for some reason?

> if you want to manually create adjacent tiles they can't be set to a common base height.
> The best solution would be an option that allows the following: that the lowest point of the tile is not [...]

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

ChHarding, 2021-01-07:
Hi,

so do you want me to implement a vertical_offset option or did the approach I outlined work for you? If you still want it, I can do it, it’s not difficult.

Cheers

Chris

> 
> Yes, I can see how defining a custom vertical datum from which to offset the base height as useful and I don’t think it would be too difficult to add this in the near future, as long as you’re OK with using it via the manual options (e.g. “vertical_offset”: 123.45).  However, I’m curious how you would know the value for all your offsets since there’s no way to get this information from the front end. Or are you just going to do all our models first, look at their elevation stats in the log, calculate the offset from that and re-run them?
> 
> But, maybe you have not tried the following which might already give you the result you’re looking for:  
> - set up you tiles in a way you want them.
> - if you need more that 8 tiles for this use ntilesx and ntilesy to set them manually (e.g. “ntilesx”:10, “ntilesy”:12)
> - this will ofc be way to much to process in one setting, so use the only manual option to only run that one tile e.g. “only”:[1,1] 
> - repeat this for all your tiles  “only”:[1,2], …,  “only”:[1,12], “only”:[2,1], …. which is granted a bit tedious but maybe less that running everything twice to get the offsets done properly. Plus, if you use the tiles, they are guaranteed to fit together perfectly, whereas in your way you’ll have to generate the tile boundary coordinates externally and type them in to be precise. (OK, maybe you don’t care about super fine precision …)
> 
> Or did you try this and it didn’t work for some reason?
> 
> 
>> if you want to manually create adjacent tiles they can't be set to a common base height.
>> The best solution would be an option that allows the following: that the lowest point of the tile is not referenced to base height but base height plus the height that this point is above the Vertical datum.
>> 
>> My goal is to print many tiles to have more or less a wall with a mountain range relief on it (lager than 1m²) and the maximum file size limit would compromise the quality drastically.
>> 
>> 
> 
> Chris Harding 
> Associate Professor
> Department of Geological & Atmospheric Sciences
> 
> TouchTerrain.geol.iastate.edu <http://touchterrain.geol.iastate.edu/>

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

closed 2021-05-05 by ChHarding