ChHarding/TouchTerrain_for_CAGEO #37: [feature-request] requirements.txt

nickvazz, 2021-01-09:
Love the flask-webapp!

Would it be possible to add your `requirements.txt` to make the installation nice and easy in a virtualenv?

Thanks!

ChHarding, 2021-01-11:
Sure! You want something like this in setup.py, right?

  'Pillow>=6.0.0',
        'earthengine-api>=0.1.232', # 1.232 is required after Aug 19, 2020
        'Flask>=1.0.2',
        'oauth2client>=4.1.3',
        'numpy>=1.17',
        'kml2geojson>=4.0.2', # for reading polygon coords from kml
        'geojson>=2.5', # for wrapping polygon data
        'defusedxml>=0.6', # safe minidom for parsing kml
        'six>=1.15.0',
# earthengine apparently uses an old version of six ...

but in a text file. Where should that requirements.txt file go in the
folder structure?
Also, I assume you have checked that for your case, pip install .
(which runs setup.py and with this the requirements part) doesn't work,
right?

Finally, be aware that I'm working on creating a docker image for the
standalone version, but I'm a docker noob and it will take a bit ...

Cheers

Chris

nickvazz, 2021-01-12:
Awesome! Thank you! 

I think the docker image will be easier than you expect 😛

ChHarding, 2021-01-12:
I’ve added a requirements.txt in the project root folder. Could you please check if that works for you?

> 
> 
> Awesome! Thank you!
> 
> I think the docker image will be easier than you expect 😛

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

closed 2021-05-05 by ChHarding