ChHarding/TouchTerrain_for_CAGEO #39: Missing example_config.json file

ansonl, 2021-02-15:
The README and standalone files mention an example_config.json file in the stuff folder for reference when running TouchTerrain_standalone.py. 

I can't find an example configuration file in json to build my custom config off of. Can an example config file in JSON format be added? Thanks.

dpy102, 2021-02-15:
You can run 'python TouchTerrain_standalone.py' and it'll generate an example_config.json.

cat example_config.json 
{
"CPU_cores_to_use": 0,
"DEM_name": "USGS/NED",
"basethick": 1,
"bllat": 44.50185267072875,
"bllon": -108.25427910156247,
"bottom_image": null,
"fileformat": "STLb",
"gpxPathHeight": 100,
"gpxPathThickness": 5,
"gpxPixelsBetweenPoints": 20,
"ignore_leq": null,
"importedDEM": null,
"importedGPX": null,
"lower_leq": null,
"max_cells_for_memory_only": 1000000,
"no_bottom": false,
"ntilesx": 1,
"ntilesy": 1,
"only": null,
"polyURL": null,
"poly_file": null,
"printres": 0.5,
"tile_centered": false,
"tilewidth": 80,
"trlat": 44.69741706507476,
"trlon": -107.97962089843747,
"unprojected": false,
"zip_file_name": "terrain",
"zscale": 1.0
}

ChHarding, 2021-02-15:
Looks like dpy102 beat me to the answer. But, I just checked standalone.py and it does say that an example json file is supposed to be in the stuff folder, so I’ll add it there. Thanks to both of you!

> 
> 
> The README and standalone files mention an example_config.json file in the stuff folder for reference when running TouchTerrain_standalone.py.
> 
> I can't find an example configuration file in json to build my custom config off of. Can an example config file in JSON format be added? Thanks.

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

closed 2021-05-05 by ChHarding