ChHarding/TouchTerrain_for_CAGEO #42: Option to create straight edges for imported DEMs

ansonl, 2021-04-04:
Is it possible to add an option to have higher resolution for side edges of the final model than specified for the rest of the model surface? 
I am importing DEMs of individual US states (USGS GMTED2010)  and would like to fit multiple printed pieces together. 
I've set the print resolution and scale to match my nozzle and get an acceptable level of detail on the surface of the model. However, downscaling the resolution leads to edges of the models becoming jagged. As seen below, the Virginia, North Carolina border which is slightly curved in the base DEMs becomes a jagged approximation. 

Thanks. 

[image]

ChHarding, 2021-04-04:
Because of the rasterization, borders that are not perfectly orthogonal will end up as some sort of jagged line.  Admittedly I just added a way to smooth out the jaggies as best as possible (http://blog.touchterrain.org/2020/11/whats-new-in-version-31.html) and that probably makes your case worse b/c it removes half quads. I could make an option to no use that optimization.

But, in your case you’re also going to suffer from different states having different UTM zones.  There are nationwide coordinate systems but the projection engine Google Earth uses doesn’t support most of the traditional continent scale projects (such as North America Lambert Conformal Conic). However, you could try WGS84 Web Mercator (EPSG 3857)  or pick a “central” UTM zone and set this for all your models. All your existing models will have their UTM zone and corresponding EPSG listed in the log.

 Finally you could go with a higher resolution e.g. a 0.2 or 0.1 mm nozzle diameter. You models will be much larger but you will get finer “jaggies”

Cheers

Chris

> Is it possible to add an option to have higher resolution for side edges of the final model than specified for the rest of the model surface?
> I am importing DEMs of individual US states (USGS [...]

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

ansonl, 2021-04-04:
TouchTerrain doesn't rely on Google if I import my own DEM right? I'm actually processing GMTED2010 in QGIS and outputting the DEM (sea level set to NODATA) in the Lambert Conformal Conic projection that you mentioned so the DEMs already processed with coordinates matching the desired print. 

I haven't looked at the TouchTerrain code in depth, it would be appreciated if you would be able to add in an option to turn off the edge optimization. 

I suppose the ultimate solution for my use case without understanding the entire TouchTerrain code base would be to:
1. Make a gigantic DEM for the entire US
2. Import it into TouchTerrain and make a gigantic STL
3. Somehow cut the STL along state borders from a shapefile. 

But I'm pretty satisfied with the efficient 80% solution of importing each state DEM into TouchTerrain with the exception of the edges at the moment.

ChHarding, 2021-04-05:
I see. I should have asked if you’re using the web or the stand alone version! As you describe it, your process is exactly how I would do it. I’ll add a no_border_opt and with this you should theoretically get matching jaggies with your method.  I’ll ping you once I’ve added that.

For non-jaggy borders you would have to use something like Blender, extrude each border polygon and clip your huge STL with it. We have a guy here who’s very good with Blender and can ask him about this if you want.

Finally:

- How tricky/smooth was it for you to get the docker container to run? (I assume you’re using docker for standalone?). I ask b/c I’m still fine tuning the installation instructions, especially for beginners or even non-coders.

- What are you using your final models for?

Cheers

Chris

> 
> 
> TouchTerrain doesn't rely on Google if I import my own DEM right? I'm actually processing GMTED2010 in QGIS and outputting the DEM (sea level set to NODATA) in the Lambert Conformal Conic projection that you mentioned so the DEMs already processed with coordinates matching the desired print.
> 
> I haven't looked at the TouchTerrain code in depth, it would be appreciated if you would be able to add in an option to turn off the edge optimization.
> 
> I suppose the ultimate solution for my use case without understanding the entire TouchTerrain code base would be to:
> 
> Make a gigantic DEM for the entire US
> Import it into TouchTerrain and make a gigantic STL
> Somehow cut the STL along state borders from a shapefile.
> But I'm pretty satisfied with the efficient 80% solution of importing each state DEM into TouchTerrain with the exception of the edges at the moment.

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

ansonl, 2021-04-06:
Awesome, that option would be really helpful! The pre-3.1 version of Touch Terrain retains all jagged edges as the no_border_opt or will the new option retain more of the jagged edges? 
Retaining matching jaggies should be sufficient for my print but I'll find out once I actually 3d print two jaggy states ;) 

I looked back at my script and I actually meant that I'm generating geotiffs in Lambert Conformal Conic and importing those into TouchTerrain standalone. 

I'm using Anaconda on Windows 10 to create a python3 environment and installed TouchTerrain in the virtual environment. I haven't used Docker.
- I'm using Spyder as the python IDE.
- I created a python script to generate the json configs and CLI commands to run TouchTerrain_standalone.py with the created GeoTIFFs.
- Then I've made a smaller python file to launch the TouchTerrain instances to process multiple states GeoTIFFs into STLs

```
from functools import partial
from multiprocessing.dummy import Pool
from subprocess import call

import datetime
first_time = datetime.datetime.now()

commands = []

with open('./touch-terrain-batch.sh', 'r') as fp:
    for line in fp:
        commands.append(line)

pool = Pool(12) # 12 concurrent commands at a time
for i, returncode in enumerate(pool.imap(partial(call, shell=True), commands)):
    if returncode != 0:
       print("%d command failed: %d" % (i, returncode))
       
later_time = datetime.datetime.now()
difference = later_time - first_time
seconds_in_day = 24 * 60 * 60
elapsed = divmod(difference.days * seconds_in_day + difference.seconds, 60)
print(str(elapsed[0]) + 'm ' + str(elapsed[1]) + 's elapsed')
```

I'm trying to print PLA terrain models of all US states and fit them together on a 8ft wide map on a wall. Then I would place circular pins at the places in the US that I've been.

ChHarding, 2021-04-09:
Anson,

I’ve added a smooth_borders option that is True by default. In your JSON configs, set “smooth_borders” to false to get the jaggies back. Note that I’d a super simple test to ensure nothing crashed but I didn’t have time to compare the effect of this flag on geotiffs with NoData. Please let me know if this works for you as expected.

Cheers

Chris

> Awesome, that option would be really helpful! The pre-3.1 version of Touch Terrain retains all jagged edges as the no_border_opt or will the new option retain more of the jagged edges?
> Retaining [...]

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

ansonl, 2021-04-11:
Chris,

Can confirm the *smooth_borders* option works. Weirdly enough, *smooth_borders* produces almost similar borders at `print_res: 0.8` as seen in the below screenshots. The smoothing is noticeable once I use a finer `print_res: 0.4`. I haven't experienced any crashing. 

Unfortunately, some complicated states borders still don't fit together well with the jagged edges. 
May I take you up on the offer to talk to the Blender guy who may have more expertise regarding exporting border polygons and cutting the mega-STL? Thank you. 

*Pre smooth_borders update*
[image] 

*smooth_borders on*
[image]

*smooth_borders off*
[image]

ChHarding, 2021-04-11:
I can ask, but I’m a bit dismayed that your non-smoothed models still don’t fit perfectly together. IIRC you’re creating masked geotiffs via QGIS. Can you put those two states you’re using in the images you sent me into QGIS and confirm that all their “teeth” properly “fit together”. If they do, I’ll need to check why my STL output doesn’t match what you’re seeing in QGIS. Otherwise, it might be easier to fix whatever is giving you those slight discrepancies rather than  jump into a whole new process.

> Chris,
>
> Can confirm the smooth_borders option works. Weirdly enough, smooth_borders produces almost similar borders at print_res: 0.8 as seen in the below screenshots. The smoothing is noticeable [...]

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

ChHarding, 2021-04-11:
Also,  can we de-couple this email chain from github? I was going to cc you on my question to Pete (the Blender guy) but the realized I don’t know how to do that with your GitHub spoofed email. What’s an “official” email I can use for that?

> 
> I can ask, but I’m a bit dismayed that your non-smoothed models still don’t fit perfectly together. IIRC you’re creating masked geotiffs via QGIS. Can you put those two states you’re using in the images you sent me into QGIS and confirm that all their “teeth” properly “fit together”. If they do, I’ll need to check why my STL output doesn’t match what you’re seeing in QGIS. Otherwise, it might be easier to fix whatever is giving you those slight discrepancies rather than  jump into a whole new process.
> 
>> 
>> 
>> Chris,
>> 
>> Can confirm the smooth_borders option works. Weirdly enough, smooth_borders produces almost similar borders at print_res: 0.8 as seen in the below screenshots. The smoothing is noticeable once I use a finer print_res: 0.4. I haven't experienced any crashing.
>> 
>> Unfortunately, some complicated states borders still don't fit together well with the jagged edges.
>> May I take you up on the offer to talk to the Blender guy who may have more expertise regarding exporting border polygons and cutting the mega-STL? Thank you.
>> 
>> Pre smooth_borders update
>>  <https://user-images.githubusercontent.com/546458/114291039-1f6a7b80-9a52-11eb-888c-b4842d3f3c84.png>
>> smooth_borders on
>>  <https://user-images.githubusercontent.com/546458/114291053-445eee80-9a52-11eb-8e70-5db2250770f2.png>
>> smooth_borders off
>>  <https://user-images.githubusercontent.com/546458/114291060-550f6480-9a52-11eb-91d6-cc051140c93a.png>
>> 
> 
> Chris Harding 
> Associate Professor
> Department of Geological & Atmospheric Sciences
> 
> TouchTerrain.geol.iastate.edu <http://touchterrain.geol.iastate.edu/>

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

ansonl, 2021-04-18:
Chris,

I sent a message to your iastate.edu email with the details. Let me know if you didn't get anything and I'll drop my email below. Thanks.

ansonl, 2021-08-06:
Following up to let any readers know that setting a configuration of `smooth_borders: False` resolves this issue of retaining straight edges for imported DEMs. Below is a 3D fit test of Maryland and Virginia borders printed in PLA. 
[image]

closed 2021-08-06 by ansonl