ChHarding/TouchTerrain_for_CAGEO #47: Output min and max elevations within generated output

doug-harriman, 2021-06-22:
On the output page that lists the parameters used for generation, I would like to request that you also output the minimum and maximum elevations within the data set.  That provides the information necessary to determine the actual Z-scaling used if the data is scaled to a fixed height.

ChHarding, 2021-06-23:
I already log the pre-z-scaled elevation min/max like this:

process started: 23:06:48.595372 

Region (lat/lon):
   46.22980441137155 -122.12893771814642 (top right)
   46.154584342537 -122.25169467848252 (bottom left) 
center at [-122.19031619831446, 46.192194376954276]  UTM 10 N ,  EPSG:32610 
lon/lat size in degrees: [0.12275696033610473, 0.0752200688345539] 
Earth Engine raster: USGS/NED 
 USGS National Elevation Dataset 1/3 arc-second 
URL for geotiff is:  https://earthengine.googleapis.com/v1alpha/projects/earthengine-legacy/thumbnails/35551c7968acbf9b6691d5a69193042e-49ddba912b6be89eef1716d27d68432d:getPixels 
 geotiff size: 0.1846637725830078 Mb 
 cell size 37.90471094952414 m, upper left corner (x/y):  557692.0122003487 5119978.831506973 
full (untiled) raster (height,width)  (224, 253) float64 elev. min/max: 795.51416015625 2536.448486328125 
cell size: 37.90471094952414 m  
adjusted print res from the requested 0.4 mm to 0.3952569169960474 mm to ensure correct model dimensions 
total model size in mm: 100.0 x 88.53754940711462 
map scale is 1 : 95898.91870229607 
From requested model height of 20.0 mm, calculated a z-scale of 1.1016948458149751 
Cells per tile (x/y) 253 x 224 
using single-core only 

1 x 1 tiles, tile size 100.00 x 88.54 mm
 
tile 1 1 : height:  1.0 - 21.0 mm , file size: 11 Mb 

total size for all tiles: 11 Mb 
added full geotiff as NED_-122.19_46.19.tif 

processing finished: 23:06:54.087557 

Granted, it’s no pretty and I could give the min/max it’s own line. Or are you looking for something different?

Cheers

Chris

> 
> 
> On the output page that lists the parameters used for generation, I would like to request that you also output the minimum and maximum elevations within the data set. That provides the information necessary to determine the actual Z-scaling used if the data is scaled to a fixed height.

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

doug-harriman, 2021-06-23:
Chris - That's what I needed.  I had not looked at the log file in the zip, my apologies.  I had only looked at the web page generated that provides the zip download link, and this info wasn't present there.  Thanks for the follow up.

closed 2021-06-23 by doug-harriman

ChHarding, 2021-06-23:
Doug,

 no problem! It’s a pretty obscure file :)  Can I ask what you’re using the app for?

Cheers

Chris

> 
> 
> Chris - That's what I needed. I had not looked at the log file in the zip, my apologies. I had only looked at the web page generated that provides the zip download link, and this info wasn't present there. Thanks for the follow up.

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

doug-harriman, 2021-06-24:
Chris -

My goal is to mill terrain profiles out of wood, then laser etch elevation contours into them.  I have a Sainsmart 3018 mill with a laser head for it.  Because there's an active Z axis unlike most laser cutters, I can run it at different focal planes.  Need the elevations to get the proper labels on the elevation lines.  I'm using Matplotlib contour plots to generate the elevation lines, then converting to G-Code. I have that working now.  Next up is to get the labels properly sized and rotated so they're aligned with the surface at the proper location.  You can check out the project [on github](https://github.com/doug-harriman/topo-box#elevation-contour-generation-tool-process-overview).

 - Doug

ChHarding, 2021-06-24:
Neat!

Looking at 

https://github.com/doug-harriman/topo-box/blob/32f45db05bcb0401126a116644ac1aa66968bc5a/oregon-black-butte/stl2topo_contours.py#L95

have you tried opening the geo tiff in GDAL? This is essentially how I read in the DEM data to the make a messy from. It’s kinda clunky and can be tricky to install but it will give you a numpy array with elevation values.

Otherwise PIL/pillow might be able to read in a geotiff. I certainly can do tiffs with float pixels but I don’t remember if I ever tried a geotiff.  

> Chris -
>
> My goal is to mill terrain profiles out of wood, then laser etch elevation contours into them. I have a Sainsmart 3018 mill with a laser head for it. Because there's an active Z axis unlike [...]

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

doug-harriman, 2021-06-24:
I haven't tried the GeoTIFF.  My initial run through this code (you can see it in the python file history) was reading in an image from https://tangrams.github.io/heightmapper/.  That was easy to deal with, but in the end, the image was too low of a resolution to deal with.  I may play with that.  Label positioning is going to be a bit of a pain with STL.  I've calculated the center of the text label, but now I need to find the mesh triangle that encloses the center point so I can determine the Z height, and the normal direction.  Prefiltering the Z-levels from the image might limit the search quite a bit.  Without some sort of optimization, it's going to be slow.

ChHarding, 2021-06-24:
https://pypi.org/project/trimesh/

say it has 

Ray-mesh queries including location, triangle index, etc. 

So I imagine you would construct a vertical ray from the bottom at the center point and it would return the triangle id?

> I haven't tried the GeoTIFF. My initial run through this code (you can see it in the python file history) was reading in an image from https://tangrams.github.io/heightmapper/ [...]

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

doug-harriman, 2021-06-24:
Awesome!  Thank you.  That's the functionality I'm after.  I'm taking tomorrow off from work, so hope to have the text G-Code generation done tomorrow.  Should see commits coming through on my project.  I also need to dink with the font size.  I need to figure out how to get the matplotlib figure set up with the correct units so that there's a clear mapping from the pt size to mm.  Always another layer to the onion.  After that, it should have all of the functionality needed, then just some code cleanup.  Hope to get milling soon.

ChHarding, 2021-06-25:
Could you use PIL/pillow instead of Matplotlib? There’s some info on how specific fonts relate to pixels:  https://stackoverflow.com/questions/43060479/how-to-get-the-font-pixel-height-using-pils-imagefont-class

> Awesome! Thank you. That's the functionality I'm after. I'm taking tomorrow off from work, so hope to have the text G-Code generation done tomorrow. Should see commits coming through on my project. [...]

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

doug-harriman, 2021-06-25:
That doesn't really solve my problem.  Matplotlib.clabel generates the text
labels for the contour elevation.  It also breaks the contour lines so that
the labels have a space to live in within the contour line.  That
functionality is important.   I need to know the size of the text in that
box so that when I generate the gcode for the text using
https://pypi.org/project/TextToGcode/ I can make it the right size to fit.

doug-harriman, 2021-07-01:
Chris -

Got my first prototype done of a topographic map with laser etched
elevation contour mapping.  It worked really well, though I have a little
work to do on the alignment of the laser image to the 3D part.   Photos at:
https://github.com/doug-harriman/topo-box#oregon-black-butte

Thanks for advice on the ray tracing with trimesh.  That worked perfectly
for both Z-height mapping and finding the surface normal to do the 3D
rotation of the text.  I also used it for taking an SVG from OpenStreetMaps
and mapping roads and trails to the proper Z-height for laser etching.  In
the end, this one is a little busy with all of those etches, but with a
little cleanup, I think it'll look really good.

 - Doug

ChHarding, 2021-07-01:
Very nice!

Couple of notes:
- if you find that the TouchTerrain model is to detailed you can always increase the nozzle size parameter, which is directly linked to the resolution of the geotiff DEM I get. I know that for CNC’lers the width in mm and nozzle size makes little sense as all the scaling happens in post processing, so I’m experimenting with CNC specific presets for both:
  Did you use these?
- Your link for mesh reduction is interesting! I personally just use MeshMixer, where I can reduce it down to X% or to Y triangles, It also does preserve sharp edges. I’m guessing that would be much faster than having Fusion do it (?)
- I like your laser etching! I had a similar idea for laser etching geologic outcrop polygons onto a 3D printed terrain model. That would be super useful for us b/c we would like to hand color the polygon according to their rock type/age. However, I blew up the cheap $50 laser I had used and now have to replace the board. I did get it to work on one example and the etching was actually strong enough to make a good line on grey PLA. Focusing, however was kinda tricky with this cheap laser. I should probably re-activate this …

Cheers

Chris 

> 
> 
> Chris -
> 
> Got my first prototype done of a topographic map with laser etched
> elevation contour mapping. It worked really well, though I have a little
> work to do on the alignment of the laser image to the 3D part. Photos at:
> https://github.com/doug-harriman/topo-box#oregon-black-butte
> 
> Thanks for advice on the ray tracing with trimesh. That worked perfectly
> for both Z-height mapping and finding the surface normal to do the 3D
> rotation of the text. I also used it for taking an SVG from OpenStreetMaps
> and mapping roads and trails to the proper Z-height for laser etching. In
> the end, this one is a little busy with all of those etches, but with a
> little cleanup, I think it'll look really good.
> 
> - Doug
> 
> 
> 
> 
> 
> > That doesn't really solve my problem. Matplotlib.clabel generates the
> > text labels for the contour elevation. It also breaks the contour lines so
> > that the labels have a space to live in within the contour line. That
> > functionality is important. I need to know the size of the text in that
> > box so that when I generate the gcode for the text using
> > https://pypi.org/project/TextToGcode/ I can make it the right size to
> > fit.
> >
> >
> >> Could you use PIL/pillow instead of Matplotlib? There’s some info on how
> >> specific fonts relate to pixels:
> >> https://stackoverflow.com/questions/43060479/how-to-get-the-font-pixel-height-using-pils-imagefont-class
> >>
> >>
> >> >
> >> >
> >> > Awesome! Thank you. That's the functionality I'm after. I'm taking
> >> tomorrow off from work, so hope to have the text G-Code generation done
> >> tomorrow. Should see commits coming through on my project. I also need to
> >> dink with the font size. I need to figure out how to get the matplotlib
> >> figure set up with the correct units so that there's a clear mapping from
> >> the pt size to mm. Always another layer to the onion. After that, it should
> >> have all of the functionality needed, then just some code cleanup. Hope to
> >> get milling soon.
> >> >
> >> https://github.com/ChHarding/TouchTerrain_for_CAGEO/issues/47#issuecomment-867962894>,
> >> >
> >>
> >> Chris Harding
> >> Associate Professor
> >> Department of Geological & Atmospheric Sciences
> >>
> >> TouchTerrain.geol.iastate.edu

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences

TouchTerrain.geol.iastate.edu

doug-harriman, 2021-07-01:
Chris -
- I thought about looking at a different nozzle diameter to see how that
impacted the mesh size, just didn't get around to it.  I'll definitely play
with that next time.  I naively went for max res when I first downloaded
the model, and never revisited.
- Thanks for the pointer on the mesh reduction.  I was doing all of the CAM
in Fusion360, and this was their recommendation.  They crap out at ~32k
triangles.  Fusion was reasonably fast, but a hassle,  The one cool feature
it has is that you can locally reduce mesh density with graphical
selection.  The downside is that there's little control of mesh reduction.
Still, all I care about is getting to the 32k triangle count so that I can
get to CAM for milling.  I'll run meshmixer next time.
- My laser was about $150 (Sainsmart 5.5W).  It has a focusable lense, but
I usually just adjust Z-height with the spindle Z axis position.  I have a
little Python script that will generate lines at different Z-heights so I
can find the one most in focus.  That's in my gcode_utilities repo, but
there are other ones out there.
- If you just want a color 3D print, HP now has color 3D printing
available.  Not as fun (or inexpensive) as doing it yourself, but it's a
simple way to get parts.  I used to work at HP, and have a good friend
there that's looking for 3D content to show off their color capabilities.
Let me know if you're interested in an introduction.

 - Doug