ChHarding/TouchTerrain_for_CAGEO #53: Local install (non Docker) causing Storage to fill up

anciltech, 2021-09-18:
I ran the "pip install ." with the setup.py and all other files in the working directory of Python, and my Mac started asking me for permission to access Photos, Contacts etc, witch I found odd, and noticed that the write speed to my local storage was very high and filling up my SSD. I terminated the code and only had a few gigs left on the drive. After a restart the storage returned back to where it was before I attempted the install. What the heck is all that about? Im new to python, but I dont think I did anything that drastically wrong. I have been working with the docker version beforehand, but wanted to see if there was any better performance in the undockerized version.

ChHarding, 2021-09-18:
Hi,

pip install . only downloads and installs the required python packages including all their prerequisites. If you’ve started with a vanilla python installation that could be a lot but you’d have seen pip commandline messages as they get installed. No sure if that would hammer your SSD or how much disk space that needs. Maybe a couple of gigs? If you started with something like Anaconda (which I do), you already have most of the large packages installed. My current Anaconda folder is about 5Gb although I have installed a lot of other packages in the last year that are not needed for TouchTerrain so I imagine a fresh Anaconda install + whatever TouchTerrain needs would be less(?)

Also, I do all my development on Win 10 b/c have actually never managed to run TouchTerrain standalone on my Mac. I never could get OsGeo/GDAL installed w/o compiling stuff for it (didn’t want to go home-brew or install a C-compiler …). 

But, just for my interest, I completely uninstalled and re-installed Docker and pulled the image:
docker pull chharding/touchterrain_jupyter
After the pull, I had about 5 Gb less disk space (2Gb download, expands to 5Gb). And that's just the docker image, the install_touchterrain.sh script will clone entire GitHub repo and then run pip install . on that. But I’m unclear if that results in even less disk space on your SSD or if that ends up “inside” the 5Gb container. Sorry, I’m really not very familiar with Docker!

Also, although I’ve never timed it properly, I don’t think I’ve noticed a big performance loss when using the docker version. Granted, that only applies to my Win 10 PC where I routinely run the non-docker version for debugging, etc. Can I ask why you’re exploring the non-docker version? Do you simply want to save the overhead disk space?  

I would recommend sticking with the docker version b/c I’m really not sure if it’s even possible to run the non-docker version on Mac b/c of the OsGeo/GDAL install issues I run into. That was ab out 1.5 years ago on an older MacOS version, so maybe that’s not an issue anymore. But I’d hate for you to waste a lot of time (like I did) on this only to end up having to fall back to docker anyway!

If you want to try it I would get rid of the docker image and also uninstall Docker.  Then start your (local) pip install . and watch the disk space in About this Mac - Storage (or maybe there’s a unix util for that?). 

Sorry, I know that’s a lot of conjecture and speculation but that’s all I can think of!  However, I can assure you that this is not some sort of crypto mining BS that gets maliciously installed :) (Although that’s ofc what any bad actor would say ….)

Final question, can I ask what you’re trying to do with TouchTerrain?

Cheers

Chris

> I ran the "pip install ." with the setup.py and all other files in the working directory of Python, and my Mac started asking me for permission to access Photos, Contacts etc, witch I found odd, and [...]

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu

anciltech, 2021-09-19:
Thanks for the detailed response Chris, it filled up about 150Gb of space so a little bit more than I think a normal install would take ;) 

I am using a resin printer to create some cool wall art and gifts, and the printer really capture every triangle face if it isn't high enough quality, ergo large file sizes and running into some of the size limits, I was looking to run the local code for increased performance and hopefully success rate. Sometimes the docker container version will slow if the files are large, until it eventually doesn't show any progress or CPU/GPU resources used; any idea why that is? 

Seems like you guys are running quite the rig behind the web version, that'll spit out a 150Gb file size in about 30 seconds where it takes quite a while on my Laptop. Maybe ill stick with that! 
Thanks for the tool, this makes things possible that I'd never be able to figure out on my own.

ChHarding, 2021-09-19:
Ah, then I have a hunch! TouchTerrain works by first processing the DEM  (this is when you see 10% …. 90%) during which it accumulates data into the tmp folder. That’s done b/c for large models this would not fit into memory, so the temp folder is just used as a sort of slow memory dump.  (BTW there’s a config setting that defines when a raster is too large to be processed fully in memory, that way small rasters can be processed pretty quickly.) TT then goes over that temp file (you’ll see a message about assembling triangles) and converts it into a STL (inside a dedicated folder) and the zips that folder.  

But, in your case something in your data or setup must lead to these bonkers large temp files. I suspect you’re doing large and/or very high resolution rasters (maybe even more than one in parallel?). How many pixels is your geotiff that you import? Or are you getting online DEM data?
What is your print resolution set to? Are you using the original raster resolution (i.e. -1) or are you setting it so a very small value (< 0.1) MY experience with SLA is limited but we should talk about what the practical limit for you case would be (i.e. below that it doesn’t give you a better print, just more processing). For typical FDM that’s around 0.4 mm (maybe down to 0.2 mmm) b/c that’s the smallest blob that a 0.4 mm nozzle can create. For example, assume you have the absolute highest peak of a steep mountain. It won’t matter that that peek is covered by 10 x 10 cells in your DEM raster, the sliver will have to amalgamate those 100 elevation values into a single value b/c that’s the size of the smallest blob that can come out of the nozzle. So it would be smarter to down-sample that DEM before processing, so that instead of 10 x 10 pixels, we’d get just 1 x 1 pixel for that peak. I suspect that that’s at the core of why you’re getting these huge files. Note that this x/y resolution has nothing to d with your fabulously small SLA layer heights! (Well, there some linkage but it’s complicated and the effect is very small :)

So what I would like you to do is to do a test run with a lower resolution or a smaller raster and watch both the output for the 2 processing stage and your tmp folder. Let’s establish if my thinking is correct, then we can talk about finding an effective print resolution for your printer, OK?

Cheers

Chris

> Thanks for the detailed response Chris, it filled up about 150Gb of space so a little bit more than I think a normal install would take ;)
>
> I am using a resin printer to create some cool wall art [...]

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu

ChHarding, 2021-10-01:
So did this resolve your issue?

> Thanks for the detailed response Chris, it filled up about 150Gb of space so a little bit more than I think a normal install would take ;)
>
> I am using a resin printer to create some cool wall art [...]

Chris Harding 
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu

anciltech, 2021-10-04:
Hi Chris, sorry for the long reply time, in the middle of moving at the moment.

Setting the "max_cells_for_memory_only"  to 10^20 definitely did help, and I'm using a 4x2 grid to utilize my 8 core processor, then combining and solidifying the models together afterwards with Meshmixer. 

I am still able to get it to get TT stuck though, when increasing the resolution further. I am using online DEM data and usually start at a .4 resolution to see how it looks, and then go as far down as I can go with the resolution before running into the TT stall issue, which typically occurs around a 0.1 and a total size of 240x180mm. I just tried setting the resolution to -1 but an error returns with "EEException: Pixel grid dimensions (14210x10725) must be less than or equal to 10000."

As far as the validity of going to such low resolutions, SLA horizontal resolution is 0.05mm and the Z can be set to 0.01mm, which in my experience means that you can really see each face of each triangle from up close, and it makes the print look "out of focus" from further back. Also, steep faces, even with high resolution, can end up coming out jagged (see attached photo of Half Dome face at a resolution of 0.2, log attached for reference). What I've found can help is the Solidify function in Meshmixer, which can help smooth out some of those artifacts while removing leftover edges and structures from the previous 4x2 grid. 

What do you think? Am I asking too much? 

[image]

[image]
[logfile.txt](https://github.com/ChHarding/TouchTerrain_for_CAGEO/files/7277987/logfile.txt)

ChHarding, 2021-10-04:
> Hi Chris, sorry for the long reply time, in the middle of moving at the moment.
>
> Setting the "max_cells_for_memory_only" to 10^20 definitely did help, and I'm using a 4x2 grid to utilize my 8 core [...]

Well, there you’re running into a Google server restriction. It simply won’t serve geotiffs that large. There are 3 ways around that:

1) Knowing the corner coordinates you can “download” very large geotiff from Google Earth Engine via a javascript tool that transfers the geotiff into your Google drive from which you can download it then manually and use it in TouchTerrain standalone. This is described at the very end of this ReadMe:  https://github.com/ChHarding/TouchTerrain_for_CAGEO#readme <https://github.com/ChHarding/TouchTerrain_for_CAGEO#readme>. 

2) Split your area into multiple tiles, say 2 x 2 = 4 tiles. Each tile should now be below the 10,000 pixel limit, if not use more tiles.
Use the only setting in Manual settings to only process one tile at a time:  E.g. “only”:[1,1] will only process tile x=1, y=1, “only”:[2,1]  for x=2, y=1, etc.
Unzip each and put the 4 stills into meshmixer (with append). They will “fit” together on the virtual build plate (i.e. touch each other to form a rectangle) and you can simple print those 4 parts together. The only downside is that you’ll still have the internal walls, so that’s a try bit of waste. Maybe Meshmixer could remove this but I never bothered to find out.  

But, there seems to be a hardware limit on our server that users can’t get over, even if they manage to get below 10k pixels. I know that b/c I now have a 400 x 400 mm printer (CR-6 max) and with 0.4 mm nozzle I can never get 400 x 400 mm through, it always times out.  Something like 380 x 380 works but not 400 x 400. According to my server guy its just a RAM limitation and we’re stuck with whatever we have for now.  So, on to ... 

3) Again using tiles, but only download the geotiffs, not the STLs and then run standalone on them. Either merge the geotiff into a single geotiff (using QGIS) or just create 4 STLs and append them into 1 model in Meshmixer.

But I would suggest that you first increase your 0.2 mm to whatever makes you lower the pixel count  below 10000 and see if that goes through. I think that would still look very nice on SLA with 0.1 mm layer height. 

Cheers

Chris

anciltech, 2021-10-06:
Hi Chris

I've done some testing over the past day on the different options above to see which would be the best...

1. Downloading TIF from EE - This also stops in the middle of processing when the file is at a similar resolution to what I've tried before. 
An interesting tangent though, pulling the TIF at the original resolution of 30m (var scale: 30 during export of DSM from JAXA/ALOS/AW3D30/V2_2) results in a far rougher model than setting the resolution to 5m, as shown in the screenshots. How is this possible if the original data is only accurate to 30m? 
[image]
[image]

2. Manually looping with "Only" - This works and seems like the best option at the moment, but is slow, since it only uses one core. Also, Meshmixer is able to remove these inner walls with the Make Solid tool, which works quite well. I am hollowing out my prints to save on resin so I have been doing this after combining. 

3. Using TT to obtain tiled TIFs - I only ran into the 10000 limit when setting the resolution to -1. I dont think Ill need full resolution at any point, but I'll look into this in the future. 

About the failure during processing, do you think it could be an issue with the multiprocessing? I was looking through your code a bit and was thinking maybe this synchronization implementation might be the trick ...
https://docs.python.org/3/library/multiprocessing.html#synchronization-between-processes

Thanks for all the help!
Max

PS Ive updated Docker and TT isn't able to start anymore. Ive deleted and reinstalled TT, and tried to Trust the notebook, but it always leads to:
[W 14:29:05.983 NotebookApp](B Notebook TouchTerrain_jupyter_for_starters.ipynb is not trusted
Operation not permitted (src/thread.cpp:309)
Operation not permitted (src/thread.cpp:309)
qemu: uncaught target signal 6 (Aborted) - core dumped
./run_touchterrain.sh: line 7:    53 Aborted                 jupyter notebook --ip 0.0.0.0 --port 8888 --no-browser --allow-root
root@a33de40e914d:/TouchTerrain# Operation not permitted (src/thread.cpp:309)
Operation not permitted (src/thread.cpp:309)
qemu: uncaught target signal 6 (Aborted) - core dumped

ChHarding, 2021-10-06:
Max,

Let’s talk about the notebook issue first.  I’ve now also upgraded to Docker 4.01 and have reinstalled TouchTerrain. This works fine for me:

It does say notebook not trusted in the CLI and in the notebook but that still lets me use the nb as usual, even the geemap cell works. I guess I don’t understand what Javascript disabled means in this context. If  I hit the Trust button it restarts the nb in trusted mode.  I guess I have no idea right now why you’re getting this: Operation not permitted (src/thread.cpp:309)

Is it possible that we have different versions of Jupyter despite using the same install script?

BTW I used the 127.0.0.1:8888 url in my browser.

Is it possible that something in your browser changed recently? What are you using?

Cheers

Chris

ChHarding, 2021-10-14:
> https://docs.python.org/3/library/multiprocessing.html#synchronization-between-processes <https://docs.python.org/3/library/multiprocessing.html#synchronization-between-processes>Unless you create tiled models the code will run in Single Core mode. If you look at log file it will say using single-code only somewhere.

Max,

Just wanted to try and answer your other points (besides the jupyter weirdness …).Please let me know if and how you manage to clear that up.

> 
> 
> Hi Chris
> 
> I've done some testing over the past day on the different options above to see which would be the best...
> 
> Downloading TIF from EE - This also stops in the middle of processing when the file is at a similar resolution to what I've tried before.

How big is that tif? And at what resolution does it stop working?

> An interesting tangent though, pulling the TIF at the original resolution of 30m (var scale: 30 during export of DSM from JAXA/ALOS/AW3D30/V2_2) results in a far rougher model than setting the [...]
So you resampled the 30 m DEM to 5 m, which is done via a linear interpolation (in this case), which tends to smooth or blur:

https://developers.google.com/earth-engine/guides/resample <https://developers.google.com/earth-engine/guides/resample> 

Note that this example shows bicubic interpolation which is even smoother than linear. If the resampling were to use nearest neighbor, your 5 m DEM would look very close to the 30 m DEM b/c multiple 5 m sample would have the exact same value as the closest 30 m sample value.

> <https://user-images.githubusercontent.com/79978026/136229973-2c196400-de82-4fe3-ac6a-fcf8c1845f1e.png> [...]
Well, so that’s not ideal but seems to give you a solution for now.