ChHarding/TouchTerrain_for_CAGEO #60: Issue while reading custom config file: importedDEM not recognized

anciltech, 2022-10-24:
**Issue:**
ImportedDEM not working with my custom config file. 

Using the same string ["importedDEM": "./stuff/pyramid.tif",] in my "custom" config file doesnt work and sets the importedDEM to None, but when using the default layout from [example_config.json] the importedDEM is used successfully. 

```
reading atiff.json

Using these config values:
...
importedDEM = None
...
```

I have moved some things around in my config files since I usually only edit a few items. I believe something about this structure breaks the importedDEM feature. 

file is structured as following: 

```
{
"zip_file_name": "attempt1",
"importedDEM": "./stuff/pyramid.tif",
"DEM_name": "JAXA/ALOS/AW3D30/V2_2",
"printres": 0.5,
"tilewidth": 275,
"ntilesx": 1,
"ntilesy": 1,

"trlat": 46.13101542129134,
"trlon": 7.839429301234198,
"bllat": 45.94432505455063,
"bllon": 7.593839091273287,

"trlat": 45.98389420078675,
"trlon": 7.669536937449504,
"bllat": 45.967530444713745,
"bllon": 7.64599071857746,

"zscale": 1,
"basethick": 1.5,

"importedGPX": [],
"gpxPathHeight": -150,
"gpxPathThickness": 4,
"gpxPixelsBetweenPoints": 1,

"CPU_cores_to_use": 0,
"bottom_image": null,
"fileformat": "STLb",
"fill_holes": null,
"ignore_leq": null,
"importedDEM": null,
"lower_leq": null,
"max_cells_for_memory_only": 100000000000000000000,
"no_bottom": false,
"offset_masks_lower": null,
"only": null,
"polyURL": null,
"poly_file": null,
"smooth_borders": true,
"tile_centered": false,
"unprojected": false
}
```

ChHarding, 2022-10-24:
I’ve not used the json config file method for stand alone from quite a while so it’s certainly possible that something broke. I’ll investigate. In the mean time, how about you give the new Jupiter colab meths a try:

http://colab.research.google.com/github/ChHarding/TouchTerrain_for_CAGEO/blob/master/TouchTerrain_jupyter_starters_colab.ipynb

This would have you upload your tiff to colab (left under File) and use it there. Note that you don’t need to get a Earth Engine dev license as you’re not using any online data. Just skip the cell that uses ee.Authenticate() and run the cell below instead.

Cheers

Chris

Chris Harding
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu<http://Touchterrain.geol.iastate.edu>

ChHarding, 2022-10-27:
Mystery solved :) You have importedDEM twice in your file so the later (with null) overwrites your first.

BTW you could also run TouchTerrain_standalone.py without a file and simply overwrite the settings in the section that says overwrite_args = …  (Line 135). Just edit the values and run the .py file. That’s how I use it myself most of the time. The json parsing is really only needed in cases where you want to batch run multiple processing runs with different settings and call TouchTerrain_standalone.py within a shell script.

Cheers

Chris

On Oct 24, 2022, at 09:34, anciltech ***@***.******@***.***>> wrote:

{
"zip_file_name": "attempt1",
"importedDEM": "./stuff/pyramid.tif",
"DEM_name": "JAXA/ALOS/AW3D30/V2_2",
"printres": 0.5,
"tilewidth": 275,
"ntilesx": 1,
"ntilesy": 1,

"trlat": 46.13101542129134,
"trlon": 7.839429301234198,
"bllat": 45.94432505455063,
"bllon": 7.593839091273287,

"trlat": 45.98389420078675,
"trlon": 7.669536937449504,
"bllat": 45.967530444713745,
"bllon": 7.64599071857746,

"zscale": 1,
"basethick": 1.5,

"importedGPX": [],
"gpxPathHeight": -150,
"gpxPathThickness": 4,
"gpxPixelsBetweenPoints": 1,

"CPU_cores_to_use": 0,
"bottom_image": null,
"fileformat": "STLb",
"fill_holes": null,
"ignore_leq": null,
"importedDEM": null,
"lower_leq": null,
"max_cells_for_memory_only": 100000000000000000000,
"no_bottom": false,
"offset_masks_lower": null,
"only": null,
"polyURL": null,
"poly_file": null,
"smooth_borders": true,
"tile_centered": false,
"unprojected": false
}

Chris Harding
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu<http://Touchterrain.geol.iastate.edu>

anciltech, 2022-10-28:
Ah man, sorry to have wasted your time with that 😬 hope it didn’t take long
ugh such a dumb mistake 

thanks for the tip, I’ll give that a try as well! I have enjoyed having the jsons so I can have a small „source“ file and I can delete 3d models to save on space when I’m done, but still have the chance to replicate the work.

closed 2022-10-28 by anciltech

ChHarding, 2022-10-28:
No worries, we’ve all been there :)