ChHarding/TouchTerrain_for_CAGEO #61: Issue with area selection on custom data of lake Constance

rs-development, 2022-11-25:
I am trying to extract a part of lake Constance, but I always get STLs of the complete lake instead of the selected area.

Maybe someone can help with choosing the correct input model and/or better defining the area. I guess that maybe the input  data is not global coordinate based or something like that.

Data is available here:

https://doi.pangaea.de/10.1594/PANGAEA.855987?format=html#download

Would appreciate any help, project is awsome anyway!

```
INFO:root:Log for creating 1 x 1 3D model tile(s) from LC_All_20151208_ETRS89_10m_ac_int.asc 
 
INFO:root:started: 14:00:02.099580 
Log for creating 1 x 1 3D model tile(s) from LC_All_20151208_ETRS89_10m_ac_int.asc 
 
started: 14:00:02.099580 
INFO:root:source raster upper left corner (x/y):  487600.0 5297000.0 
INFO:root:source raster cells size 10.0 m  (4000, 6990) 
INFO:root:projection:  
INFO:root:z-scale: 1 
INFO:root:basethickness: 3 
INFO:root:fileformat: STLb 
INFO:root:tile_centered: False 
INFO:root:no_bottom: False 
INFO:root:no_normals: True 
INFO:root:ignore_leq: None 
INFO:root:lower_leq: None 
INFO:root:importedGPX: [] 
INFO:root:undefined GDAL value: -9999.0 
source raster upper left corner (x/y):  487600.0 5297000.0 
source raster cells size 10.0 m  (4000, 6990) 
projection:  
z-scale: 1 
basethickness: 3 
fileformat: STLb 
tile_centered: False 
no_bottom: False 
no_normals: True 
ignore_leq: None 
lower_leq: None 
importedGPX: [] 
undefined GDAL value: -9999.0 
INFO:root:tile_width: 750 
INFO:root:tile_height: 429.1845493562232 
INFO:root:source raster width 69900.0 m, cell size: 10.0 m, elev. min/max is 190.63699340820312 611.9719848632812 m 
INFO:root:source raster 3D print resolution would be 0.1072961373390558 mm 
tile_width: 750 
tile_height: 429.1845493562232 
source raster width 69900.0 m, cell size: 10.0 m, elev. min/max is 190.63699340820312 611.9719848632812 m 
source raster 3D print resolution would be 0.1072961373390558 mm 
INFO:root:re-sampling LC_All_20151208_ETRS89_10m_ac_int.asc :
  (6990, 4000) 0.1072961373390558 mm  10.0 m  190.63699340820312 - 611.9719848632812 m to 
re-sampling LC_All_20151208_ETRS89_10m_ac_int.asc :
  (6990, 4000) 0.1072961373390558 mm  10.0 m  190.63699340820312 - 611.9719848632812 m to 
INFO:root:  (1875, 1072) 0.4 mm  37.28 m  190.63815 - 583.76794 m 
INFO:root:print res is 0.4 mm 
INFO:root:map scale is 1 : 93200.0 
INFO:root:Cells per tile (x/y) 1875 x 1072 
INFO:root:using single-core only 
  (1875, 1072) 0.4 mm  37.28 m  190.63815 - 583.76794 m 
print res is 0.4 mm 
map scale is 1 : 93200.0 
elev min/max : 190.64 to 583.77
Cells per tile (x/y) 1875 x 1072 
using single-core only 
processing tile: 1 1
top min/max: 3.0 7.2181306
creating internal triangle data structure for <_MainProcess name='MainProcess' parent=None started>
10 % <_MainProcess name='MainProcess' parent=None started>
20 % <_MainProcess name='MainProcess' parent=None started>
30 % <_MainProcess name='MainProcess' parent=None started>
40 % <_MainProcess name='MainProcess' parent=None started>
50 % <_MainProcess name='MainProcess' parent=None started>
60 % <_MainProcess name='MainProcess' parent=None started>
70 % <_MainProcess name='MainProcess' parent=None started>
80 % <_MainProcess name='MainProcess' parent=None started>
90 % <_MainProcess name='MainProcess' parent=None started>
100% <_MainProcess name='MainProcess' parent=None started> 

Writing tile into temp. file /TouchTerrain/standalone/tmp/myterrain11.tmp
assembling binary stl from 1779772 triangles  tmp/myterrain11.tmp
10 %, 20 %, 30 %, 40 %, 50 %, 60 %, 70 %, 80 %, 90 %, 100 %, 

tile 1 1 STLb 84.8662223815918 Mb 
INFO:root:
1 x 1 tiles, tile size 750.00 x 428.80 mm
 
INFO:root:tile 1 1 : height:  3.0 - 7.2181306 mm , file size: 85 Mb 
INFO:root:
total size for all tiles: 85 Mb 
INFO:root:
processing finished: 14:00:53.361382 

1 x 1 tiles, tile size 750.00 x 428.80 mm
 
tile 1 1 : height:  3.0 - 7.2181306 mm , file size: 85 Mb 

total size for all tiles: 85 Mb 
zip finished: 14:00:53.361324

processing finished: 14:00:53.361382 

DONE!

Created zip file tmp/myterrain.zip 84.87 Mb
´´´

ChHarding, 2022-12-02:
Hi!

I’m sorry,  I should have answered this earlier! So I d/l IGKB-Tiefenschaerfe-Bodensee_DGM-10m_ETRS89-UTM32N_20151208.zip<https://store.pangaea.de/Publications/Wessels-etal_2016/IGKB-Tiefenschaerfe-Bodensee_DGM-10m_ETRS89-UTM32N_20151208.zip> and loaded LC_All_20151208_ETRS89_10m_ac_int.asc
 Into ArcGIS. I assume when you say the complete lake you’re saying that the -9999.0 values do not get ignored? Here’s how it looks in ArcGIS with NoData (-9999.0) cells in pink:

[cid:4D10D7AB-CE02-49E6-84A9-B790699AA184]

So does your STL include to pink areas? In theory setting the undefined GDAL value to -9999.0 should have removed these cells but if that’s not working you could always use the ignore_leq setting in your code like this:

args[“ignore_leq”] = -1000 (or something much lower than you actual true lowest elevation of 190.637), which should convert -9999.0 into proper undefined values.

Cheers

Chris

https://doi.pangaea.de/10.1594/PANGAEA.855987?format=html#download

Chris Harding
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu<http://Touchterrain.geol.iastate.edu>

rs-development, 2022-12-06:
Thank you for responding.

I can not see your image, maybe it got lost in the Mail to Github Comment conversion.

What i mean by the complete lake is that the resulting STL file contains the terrain of the whole lake, see image.
[image]

But the coordinates should limit that to the upper left part.

ChHarding, 2022-12-06:
Oh, I see!  Well, when using a raster DEM file, any area coordinates are ignored (as is a kms file). Reason is that the coordinates are only used to download an area from an online DEM, but you already have a file DEM. I guess I never though of anybody wanting to use the coordinates to clip a file DEM. I could implement this eventually  but for now you’ll have to clip your raster externally (e.g. with QGIS) and use that with stand alone.

Load the .asc file into QGIS, go to raster - Clip Raster by Extend. Either type in the clipping extend (you can see the UTM coordinates of your mouse cursor at the bottom of the map) or click on the … and use Select Extend on Canvas (= Map). Make a red box, set an output file name  (will be a tif) and hit run.  Then use the tif with Touchterrain standalone.

Here’s my eye-balled attempt:  https://iastate.box.com/s/y38pxllvtu3wrb5blkuh02142xmjrra9

Cheers

Chris

[image]<https://user-images.githubusercontent.com/35149229/206017664-9657e7b2-5f71-4944-97c9-4a9320563396.png>

ChHarding, 2022-12-06:
> I just came back here after going over the docs again and just saw exactly that.
>
> Thanks for the help!

You’re welcome!  I’ve also a clarification about this, to the colab notebook. So can I ask what you’re using the model for?

rs-development, 2022-12-08:
Sure. A friend of mine is working for the administration of lake constance. They have a terrain model of the lake made out of wood, which is multiple square meters in size. I'd like to 3D-Print a more handy version for day to day purposes. The wook model is most of the time stored away and only used for bigger events.

ChHarding, 2022-12-08:
Oh that’s cool! I’ve done something like that with Crater lake in Oregon, where I combined the DEM with the bathymetry. Were you planning to do something similar?

rs-development, 2022-12-14:
Actually yes. But I am still split minded about using a z_scale factor or not and if how much for the lake and how much the surroundings. I will show you our results once done. Hopefully before christmas.

ChHarding, 2022-12-14:
My experience from a purely 3D printing perspective is that you want your model to be so tall, that you can have at least 100 layers. So if you were to print at 0.2 mm layerheight, I would recommend z-scaling the model so it’s at least 2 cm tall. Also, from having printed many very large (40 x 40 cm) model with ludicrous z-scales (as Iowa is pretty flat :)such as x2 or x3, I don’t think many people would find that super unrealistic, i.e. it doesn’t seem to take away from the experience and is in fact helpful in sort of magnifying details. But I guess it depends what the official museum stance is on this.