ChHarding/TouchTerrain_for_CAGEO #63: Applying Rotation to Selection

keerl, 2022-12-02:
Is it possible to apply a rotation to a selection? I am able to do this manually right now by creating a polygon (that is just a rotated rectangle), but the edges of the STL are all jagged when I do that (see attached image).

Any ideas on how I can get around this?

[image]

ChHarding, 2022-12-02:
Karl,

You’re getting these jaggies b/c the mesh is based on a raster/grid, so any non-orthogonal (perfectly east-west or north-south) border will look like this. Note that I’ve already incorporated code to make it smoother: http://blog.touchterrain.org/2020/11/whats-new-in-version-31.html

To get truly smooth side you will have to postprocess your STL with something like MeshMixer which has a clipping plane:
[cid:94EBEAE1-45F3-4645-8C52-FAD8E6914614]
[cid:CB5E7644-FFDF-4C46-85CD-2979143FE6F1]

If you need a really precise box (with perfect 90 degree angles) you can create a template box in Meshmixer and perform a boolean intersection:
[cid:33DF2A70-1CC1-4356-BC0A-C0700FA368FA]
[cid:BB41E224-2A85-4E5B-BBB4-BBAE7D799143]

[Screen Shot 2022-12-02 at 9 05 46 AM]<https://user-images.githubusercontent.com/24727492/205335706-f3bc02ed-83ce-4c76-a0cd-a58dea958227.png>

Chris Harding
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu<http://Touchterrain.geol.iastate.edu>

keerl, 2022-12-02:
Thanks for the quick response! That makes perfect sense. Thank you.

closed 2022-12-02 by keerl

ChHarding, 2022-12-02:
You’re welcome! Can I ask you what you’re printing and what for? (Just curious …)

keerl, 2022-12-02:
I want to CNC mill out a mountain range for an art project just for fun. The rotation comes into play because the mountain range goes at a diagonal and I'd like to have the mountains only and not the flat stuff around it. Before and after examples attached below.

[image]
[image]

ChHarding, 2022-12-02:
Oh cool! I’ve never done any CNC myself but I do know that about 20% of downloaders use TouchTerrain for this. Coincidently, the author of this: https://www.instructables.com/Carving-3D-Terrain-Topographic-Relief-Maps-on-a-CN/ just contacted me.

[Screen Shot 2022-12-02 at 11 25 49 AM]<https://user-images.githubusercontent.com/24727492/205360838-117c971f-d805-4f5a-b662-ea15fcc21612.png>

[Screen Shot 2022-12-02 at 11 26 14 AM]<https://user-images.githubusercontent.com/24727492/205360836-a24acb7c-6390-4f25-831d-03783f79e498.png>