ChHarding/TouchTerrain_for_CAGEO #64: Possible Invalid STL File

keerl, 2022-12-05:
I am trying to import an STL into a program called [PyCAM](https://pycam.sourceforge.net/). When I import it I get many errors about invalid triangles:

`Skipping invalid triangle: (160.36862182617188, 6.439293542825908e-15, 0.0) / (160.36862182617188, 6.439293542825908e-15, 39.96474075317383) / (158.62548828125, 6.439293542825908e-15, 40.60410690307617)`
[many more...]

`Inconsistent normal/vertices found in facet definition 21833 of '/pycam/tmptc037vdc.stl'. Please validate the STL file!`

The model ends up loading but is missing the edges of the STL. The top mesh seems fine. It seems like data connecting the bottom rectangle data to the top mesh is corrupt possibly?

[image]

The STL seems to load fine in other programs so I'm not sure if most programs have built-in validation and PyCAM does not.

Any ideas?

ChHarding, 2022-12-05:
Karl, this is b/c the (flat) bottom consist of only 32 triangles to optimize the triangle count (which can delimited be our server quota). On the API side I think there’s a way to force it to have the same number to triangles on the bottom as on the top but I have to look if and how that would be exposed to the web app. If needed it’s easy to put in an override flag. I’ll get back to you.

[Screenshot 2022-12-05 at 7 31 25 AM]<https://user-images.githubusercontent.com/24727492/205669258-cc56b585-bb6b-4c84-ab46-dcc81920bd33.png>

Chris Harding
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu<http://Touchterrain.geol.iastate.edu>

ChHarding, 2022-12-06:
Just looked at the code and it’s not trivial to add an option to force a “full” bottom. Do you actually need the bottom? If not you could use “no_bottom”:true ….

keerl, 2022-12-08:
When I try to enable `no_bottom` I get this error: `'NoneType' object has no attribute 'get_copy'`

ChHarding, 2022-12-08:
Hmm, can I get more details on the error? Or how tricky would it be for me to reproduce your workflow so I can run it in my debugger?

keerl, 2022-12-10:
I am unable to recreate the issue now. It was likely caused by something else in my code.

closed 2022-12-10 by keerl

keerl, 2022-12-12:
I actually ended up switching to another software (for unrelated reasons) and the problem doesn't exist there.