ChHarding/TouchTerrain_for_CAGEO #69: Error Message when using AW3D30

Topographer123, 2023-02-18:
Hello guys,

not sure if this is the right place to ask but whenever I try to create a model using the AW3D30 database, I get the following error message:

Error: ImageCollection.load: Expected asset 'JAXA/ALOS/AW3D30/V2_2' to be an ImageCollection, found 'Image'.

The error does not appear when I change back to USGS and leave all the other parameters the same.

Does someone know how to fix this? Thank you!

ChHarding, 2023-02-19:
Hi,

No this is totally my fault! There’s a new version of AW3D30 that I played around with and this introduced this bug, thanks for catching it. I’ve (hopefully) fixed it and am currently rebuilding the server so you should not see this issue anymore.

Chris Harding
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu<http://Touchterrain.geol.iastate.edu>

Topographer123, 2023-02-19:
Fantastic, thank you!

ChHarding, 2023-04-10:
fixed