ChHarding/TouchTerrain_for_CAGEO #71: How to get a consistent scale across multiple polygon maps?

tueboesen, 2023-03-05:
I am trying to combine Touch Terrain with geo-countries, (which gives me Polygons and MultiPolygons for all countries in the world). My overall hobby project is essentially to combine these two in order to produce maps sorta similar to [this](https://i.etsystatic.com/10250157/r/il/ea8cdc/3520666568/il_1140xN.3520666568_1h46.jpg).

One conceptual issue I am having though is how do I get a consistent scaling of all countries?
Unfortunately it does not seem like there is a scaling parameter from what I can see. I should be able to get a similar result by using tilewidth, but I am unsure how exactly bllat, bllon, trlat, and trlon gets calculated for a polygon and what exactly tilewidth is in such a case? 

Cheers

ChHarding, 2023-03-05:
There may be a solution in the near future. Another user requested an option to set the (xy) scale manually instead of setting a build area width/height. I’ve looked at that briefly and it should not be too complicated to do, eventually …

BTW if you do this you will need to override the projection (see projection parameter here: https://github.com/ChHarding/TouchTerrain_for_CAGEO#processing-parameters)

Setting it to Web Mercator (EPSG 3857) will horrendously distort towards the poles but that may not be your concern. Better world wide projections, Like Robinson, exist but I don’t know if Google Earth Engine supports those …)

I’ll keep you posted ...

Chris Harding
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu<http://Touchterrain.geol.iastate.edu>

tueboesen, 2023-03-06:
Okay I am not in any hurry with my project so I will keep an eye on the development of this project for now. 
Thank you very much for developing all of this and making it so easily available.

ChHarding, 2023-04-18:
I've implemented a tilewidth_scale option that calculates a new tilewidth (i.e. buildplate width) from that (in mm). I tested it with online data and it seems to work, but the final scale is a bit off (I requested 1:100,000 and the final output was 1:104,000) This has to do with trying to compensate for distortions but the compensation is pretty simplistic. For your case you should definitively use projection=3857  (Web Mercator), otherwise your pieces won't fit. You should test this with adjacent countries that have a large latitude difference like China's border with Mongolia and with Mynamar. BTW feedback from a similar project (printing US states) suggests that you will probably have to shrink the STLs a bit otherwise the fit is too tight. You could do this with something like meshmixer or, if you use Cura you can set a negative value for horizontal expansion (maybe 0.5 mm?) https://the3dprinterbee.com/cura-horizontal-expansion/ Let me know how things turn out!

tueboesen, 2023-04-19:
This looks very promising, I will let you know how it turns out.

closed 2023-04-19 by tueboesen