ChHarding/TouchTerrain_for_CAGEO #72: Polygon failure at Antarctic

tueboesen, 2023-03-05:
I encounter the following issue when trying to create a stl file for an area in the Antarctic using polygons. I tried switching over to the USGS/GMTED2010 map, since it seems like the error complains about the polygon being outside the map, but even with USGS/GMTED2010 which should be global I am getting this issue.

The polygon coordinates are from https://datahub.io/core/geo-countries, but there could of course easily be a problem with them as well. I have written out the coordinates explicitly for the problematic polygon at the end of the issue in case that helps determining why this fails.

```

Log for creating 3D model tile(s) for  GMTED2010_-162.20_-81.43 
 
DEM_name = USGS/GMTED2010 
trlat = -81.26699187 
trlon = -160.37812728 
bllat = -81.60217713 
bllon = -164.02582272 
printres = 0.4 
ntilesx = 1 
ntilesy = 1 
tilewidth = 120 
basethick = 0 
zscale = 2.0 
fileformat = STLb 
no_bottom = False 
unprojected = False 
no_normals = True 

process started: 15:49:54.104320 

Region (lat/lon):
   -81.26699187 -160.37812728 (top right)
   -81.60217713 -164.02582272 (bottom left) 
Too far south for UTM - using Antarctic Polar Stereographic projection (EPSG 3031) 
lon/lat size in degrees: [3.647695440000007, 0.3351852600000029] 
requesting 202.25598122260257 m resolution from EarthEngine

INFO:root:Earth Engine raster: USGS/GMTED2010 
INFO:root: GMTED2010: Global Multi-resolution Terrain Elevation Data 2010 

Earth Engine raster: USGS/GMTED2010 
 GMTED2010: Global Multi-resolution Terrain Elevation Data 2010 

INFO:root:URL for geotiff is:  https://earthengine.googleapis.com/v1alpha/projects/earthengine-legacy/thumbnails/cb64ca3d581a0439a2faf8d47efdb426-ec6bf552d7e0be4a8fc702c7e5026fcd:getPixels 

URL for geotiff is:  https://earthengine.googleapis.com/v1alpha/projects/earthengine-legacy/thumbnails/cb64ca3d581a0439a2faf8d47efdb426-ec6bf552d7e0be4a8fc702c7e5026fcd:getPixels 

INFO:root: geotiff size: 0.0007810592651367188 Mb 
INFO:root: cell size 202.25598122260257 m, upper left corner (x/y):  -307631.3474395785 -862014.9919707321 

 geotiff size: 0.0007810592651367188 Mb 
 cell size 202.25598122260257 m, upper left corner (x/y):  -307631.3474395785 -862014.9919707321 

---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
Cell In[30], line 40
     37             print(f"{outname}: {A:2.2e}")
     38             if A > minarea: 
     39                 #try:
---> 40                 totalsize, full_zip_file_name = TouchTerrain.get_zipped_tiles(**args)
     41                 print("Created zip file", full_zip_file_name, "%.2f" % totalsize, "Mb")
     42                 #except:
     43                 #    print(f"Failure at: {outname}: {A:2.2e}")
     44                 #    print(polygon)
     45                 #    failure = True
     46 #                    break
     47     else:

File /TouchTerrain/standalone/touchterrain/common/TouchTerrainEarthEngine.py:896, in get_zipped_tiles(DEM_name, trlat, trlon, bllat, bllon, polygon, polyURL, poly_file, importedDEM, printres, ntilesx, ntilesy, tilewidth, basethick, zscale, fileformat, tile_centered, CPU_cores_to_use, max_cells_for_memory_only, temp_folder, zip_file_name, no_bottom, bottom_image, ignore_leq, lower_leq, unprojected, only, original_query_string, no_normals, projection, use_geo_coords, importedGPX, gpxPathHeight, gpxPixelsBetweenPoints, gpxPathThickness, map_img_filename, smooth_borders, offset_masks_lower, fill_holes, min_elev, **otherargs)
    894     s += "For the web app, ensure that your red selection box is at least partially covered by the grey hillshade overlay "
    895     s += "or try using AW3D30 as DEM source."
--> 896     assert False, s # bail out
    898 # For AU/GA/AUSTRALIA_5M_DEM, replace all exact 0 value with NaN
    899 # b/c there are spots on land that have no pixels, but these are encoded as 0 and 
    900 # need to be marked as NaN otherwise they screw up the thickness of the base
    901 if DEM_name == "AU/GA/AUSTRALIA_5M_DEM":

AssertionError: All(!) elevation values are -32768.0! This may happen if the DEM source does not cover the selected area.
For the web app, ensure that your red selection box is at least partially covered by the grey hillshade overlay or try using AW3D30 as DEM source.

```

`{"coordinates": [[[-160.48766, -81.597101], [-160.571116, -81.598891], [-161.022171, -81.582138], [-161.473227, -81.565385], [-161.924283, -81.548633], [-162.375338, -81.53188], [-162.826394, -81.515127], [-163.277449, -81.498375], [-163.728505, -81.481622], [-163.753163, -81.470473], [-163.748606, -81.468683], [-163.67101, -81.465427], [-163.658193, -81.461602], [-163.670155, -81.45436], [-163.948842, -81.418878], [-163.990061, -81.403497], [-163.969838, -81.393162], [-163.949208, -81.378351], [-163.932118, -81.361017], [-163.922678, -81.343357], [-163.851186, -81.318943], [-163.417531, -81.29461], [-162.983876, -81.270278], [-162.522247, -81.308787], [-162.060618, -81.347296], [-161.598988, -81.385805], [-161.137359, -81.424314], [-160.67573, -81.462823], [-160.632273, -81.490899], [-160.628529, -81.496759], [-160.671498, -81.507989], [-160.685129, -81.514744], [-160.647206, -81.517836], [-160.442209, -81.560235], [-160.413889, -81.573663], [-160.445383, -81.590509], [-160.48766, -81.597101]]], "type": "Polygon"}`

ChHarding, 2023-03-05:
This might be an issue with the polar data set, maybe with the projection it switches to by default. In fact for your world map, you can’t use it b/c it would make the poles look something like this:
[cid:A16DE8B7-77F5-476F-94E4-B58DE47811F9]

Can you try it again but not set the projection to 3857. Do you know how to do that?

Chris Harding
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu<http://Touchterrain.geol.iastate.edu>

tueboesen, 2023-03-06:
I see that you have args['projection'] which I am guessing is how to set the projection, however I haven't set that at all, so it is selecting the projection automatically thus far, and based on what I can see in the logfiles I don't think it is choosing the 3857 projection that you are telling me to avoid: 
`Too far south for UTM - using Antarctic Polar Stereographic projection (EPSG 3031)) `
So it seems like it is using 3031?

ChHarding, 2023-03-06:
Sorry, silly typo, I meant:

Can you try it again but NOW set the projection to 3857

tueboesen, 2023-03-07:
That unfortunately didn't seem to work.

```
Log for creating 3D model tile(s) for  GMTED2010_-162.20_-81.43 
 
DEM_name = USGS/GMTED2010 
trlat = -81.26699187 
trlon = -160.37812728 
bllat = -81.60217713 
bllon = -164.02582272 
printres = 0.4 
ntilesx = 1 
ntilesy = 1 
tilewidth = 120 
basethick = 0 
zscale = 1.5 
fileformat = STLb 
no_bottom = False 
unprojected = False 
no_normals = True 
projection = 3857 

process started: 06:29:13.539837 

Region (lat/lon):
   -81.26699187 -160.37812728 (top right)
   -81.60217713 -164.02582272 (bottom left) 
using EPSG:3857 as projection 
lon/lat size in degrees: [3.647695440000007, 0.3351852600000029] 
requesting 202.25598122260257 m resolution from EarthEngine

INFO:root:Earth Engine raster: USGS/GMTED2010 
INFO:root: GMTED2010: Global Multi-resolution Terrain Elevation Data 2010 

Earth Engine raster: USGS/GMTED2010 
 GMTED2010: Global Multi-resolution Terrain Elevation Data 2010 

INFO:root:URL for geotiff is:  https://earthengine.googleapis.com/v1alpha/projects/earthengine-legacy/thumbnails/e3bd37b01bbb1575499879b747f7bc91-e328801e3af9132b1ac13708ae257645:getPixels 

URL for geotiff is:  https://earthengine.googleapis.com/v1alpha/projects/earthengine-legacy/thumbnails/e3bd37b01bbb1575499879b747f7bc91-e328801e3af9132b1ac13708ae257645:getPixels 

INFO:root: geotiff size: 0.008158683776855469 Mb 
INFO:root: cell size 202.25598122260257 m, upper left corner (x/y):  -18255422.609170884 -16409027.756589746 

 geotiff size: 0.008158683776855469 Mb 
 cell size 202.25598122260257 m, upper left corner (x/y):  -18255422.609170884 -16409027.756589746 

---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
Cell In[17], line 41
     38             print(f"{outname}: {A:2.2e}")
     39             if A > minarea: 
     40                 #try:
---> 41                 totalsize, full_zip_file_name = TouchTerrain.get_zipped_tiles(**args)
     42                 print("Created zip file", full_zip_file_name, "%.2f" % totalsize, "Mb")
     43                 #except:
     44                 #    print(f"Failure at: {outname}: {A:2.2e}")
     45                 #    print(polygon)
     46                 #    failure = True
     47 #                    break
     48     else:

File /TouchTerrain/standalone/touchterrain/common/TouchTerrainEarthEngine.py:896, in get_zipped_tiles(DEM_name, trlat, trlon, bllat, bllon, polygon, polyURL, poly_file, importedDEM, printres, ntilesx, ntilesy, tilewidth, basethick, zscale, fileformat, tile_centered, CPU_cores_to_use, max_cells_for_memory_only, temp_folder, zip_file_name, no_bottom, bottom_image, ignore_leq, lower_leq, unprojected, only, original_query_string, no_normals, projection, use_geo_coords, importedGPX, gpxPathHeight, gpxPixelsBetweenPoints, gpxPathThickness, map_img_filename, smooth_borders, offset_masks_lower, fill_holes, min_elev, **otherargs)
    894     s += "For the web app, ensure that your red selection box is at least partially covered by the grey hillshade overlay "
    895     s += "or try using AW3D30 as DEM source."
--> 896     assert False, s # bail out
    898 # For AU/GA/AUSTRALIA_5M_DEM, replace all exact 0 value with NaN
    899 # b/c there are spots on land that have no pixels, but these are encoded as 0 and 
    900 # need to be marked as NaN otherwise they screw up the thickness of the base
    901 if DEM_name == "AU/GA/AUSTRALIA_5M_DEM":

AssertionError: All(!) elevation values are -32768.0! This may happen if the DEM source does not cover the selected area.
For the web app, ensure that your red selection box is at least partially covered by the grey hillshade overlay or try using AW3D30 as DEM source.

```

ChHarding, 2023-03-07:
Alright, I’ll reproduce it on my side and will hopefully figure out what’s wrong.  You’re still using this as your polygon, right?

{"coordinates": [[[-160.48766, -81.597101], [-160.571116, -81.598891], [-161.022171, -81.582138], [-161.473227, -81.565385], [-161.924283, -81.548633], [-162.375338, -81.53188], [-162.826394, -81.515127], [-163.277449, -81.498375], [-163.728505, -81.481622], [-163.753163, -81.470473], [-163.748606, -81.468683], [-163.67101, -81.465427], [-163.658193, -81.461602], [-163.670155, -81.45436], [-163.948842, -81.418878], [-163.990061, -81.403497], [-163.969838, -81.393162], [-163.949208, -81.378351], [-163.932118, -81.361017], [-163.922678, -81.343357], [-163.851186, -81.318943], [-163.417531, -81.29461], [-162.983876, -81.270278], [-162.522247, -81.308787], [-162.060618, -81.347296], [-161.598988, -81.385805], [-161.137359, -81.424314], [-160.67573, -81.462823], [-160.632273, -81.490899], [-160.628529, -81.496759], [-160.671498, -81.507989], [-160.685129, -81.514744], [-160.647206, -81.517836], [-160.442209, -81.560235], [-160.413889, -81.573663], [-160.445383, -81.590509], [-160.48766, -81.597101]]], "type": "Polygon"}

tueboesen, 2023-03-07:
Yes that is the one. Thank you for looking into it.

ChHarding, 2023-04-07:
Sorry for the delay!  The DEM source you're using doesn't actually cover the world that much South, so that's why you were getting the out-of range error. I just tried your polygon with NOAA/NGDC/ETOPO1 instead of USGS/GMTED2010 and got this. It's something but I don't know what :)

[image]

I then digitized this area in Antarctica:
[image]

which seems to work fine:
[image]

(Obviously your get an enormous distortion that close to a pole!) So I don't know what to tell you re your world map project. My solution would be to just not print Antarctica as it'd be ludicrously distorted anyway. Alternatively I could help you to get it as a polar projection which would look like this:
[image]

tueboesen, 2023-04-09:
Thank you so much for this!

I think I will start without Antarctica and then return to that once I have everything else sorted out. Though I will probably have to figure something out with Antarctica since my girlfriend studies glaciers, so a world map without Antarctica will probably not be acceptable.

closed 2023-04-09 by tueboesen