ChHarding/TouchTerrain_for_CAGEO #73: Request to implement importedGPX function to web app

AliasRuediger, 2023-03-08:
Is there a a way to use the importedGPX functionality within the web app, e.g. by using the command line/ manual settings which is already available?

Cheers!

ChHarding, 2023-03-08:
No,  b/c there’s currently no way to upload a gpx file to the server.

But you could do it in the standalone version e.g. in Colab http://colab.research.google.com/github/ChHarding/TouchTerrain_for_CAGEO/blob/master/TouchTerrain_jupyter_starters_colab.ipynb  where you can manually upload files.

Chris Harding
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu<http://Touchterrain.geol.iastate.edu>

theonlytruth, 2026-09-07:
I just want to emphasize this wish. Are there plans to include this feature in the web version?

ChHarding, 2026-09-09:
To be perfectly honest, right now I really can't spare the time b/c of a very large  teaching load. I will have  a look during the Thanksgiving break.  Until then, here's a gpx -> kml converter: https://gpx2kml.com/

theonlytruth, 2026-09-14:
> To be perfectly honest, right now I really can't spare the time b/c of a very large  teaching load. I will have  a look during the Thanksgiving break.  Until then, here's a gpx -> kml converter: https://gpx2kml.com/
> […](#)

Thanks a lot for taking the time for the answer. Don't hurry. It is not urgent.
But can just explain what you mean with the gpx to kml converter? What should I do with the kml file I would get?

ChHarding, 2026-09-16:
Sorry, brainfart ;) I was assuming that you were using a polygon and wanted to use that to clip your print area. But I now remember that gpx data is line data meant for incorporating path into the print. Sorry!