ChHarding/TouchTerrain_for_CAGEO #74: Standalone Python Library only using one core

MrMarlin, 2023-03-12:
Hi at all

Firstly thank you for developing and maintaining such a great project. I am really enjoying it.
My issue is, that although I set the `CPU_cores_to_use` argument to 8 the script is still using only one core. It also logs `using single-core only`.

Is there a setting I am missing or what would be the steps to enable the script to use more than one core?

Thanks and have a nice one

ChHarding, 2023-03-13:
Sebastian,

A while ago our ISU servers started to have issues when the app was using multi cores (typically b/c users wanted to process tiles) and we just didn’t have the time to figure out what was wrong with it. I should say that multi-processing is very much at the periphery of my Python programming skill. So the solution was to hardcore if to 1 core in TouchTerrainEarthEngine.py:516 https://github.com/ChHarding/TouchTerrain_for_CAGEO/blob/eb026d3590b57bf48eed457aa8367e01d5a22398/touchterrain/common/TouchTerrainEarthEngine.py#L516

Now, I just removed that to see if it wold work inside a jupyter notebook but I’m also getting weird errors that I can’t even debug b/c of running in multi processing mode. So I don’t really know how to even approach this :(

Is this something you have expertise in?

Cheers

Chris

Chris Harding
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu<http://Touchterrain.geol.iastate.edu>

MrMarlin, 2023-03-14:
Hi Chris

Thank you for your detailed answer. I thought the multi-core functionality was only disabled for the server. My bad. 

Unfortunately I am by no means experienced with multi processing. But if I find some time to peek my nose into it I will give it a go. Maybe I find a quick win.

cheers
Sebastian

ChHarding, 2023-03-14:
If I get some time I’ll try to isolate a minimal case.  BTW the newest version on Github should default to single core and any server version (like ours) should not have that ability anymore to switch to to multi core via user parameters.