ChHarding/TouchTerrain_for_CAGEO #77: Misaligned Tiles

lukezirngibl, 2023-04-03:
Hey Chris,

I seem to be getting misaligned tiles when generating.

[image]
[image]

My config:

```
          "trlat": 47.88266953319475,# bottom left corner lat
          "trlon": 10.707753906131003,# bottom left corner long
          "bllat": 45.62561851186426, #top right corner lat
          "bllon": 5.813401946902514,# top right corner long
          "importedDEM": "./data.tif", # if not None, the raster file to use as DEM instead of using GEE (null in JSON)
          "printres": 0.5,  # resolution (horizontal) of 3D printer (= size of one pixel) in mm
          "ntilesx": 9,      # number of tiles in x and y
          "ntilesy": 6,
          "tilewidth": 100, # width of each tile in mm (<- !!!!!), tile height is calculated
          "basethick": 15, # thickness (in mm) of printed base
          "zscale": 1.0,      # elevation (vertical) scaling
          "fileformat": "STLb",  # format of 3D model files: "obj" wavefront obj (ascii),"STLa" ascii STL or "STLb" binary STL
          "tile_centered": False, # True-> all tiles are centered around 0/0, False, all tiles "fit together"
          "zip_file_name": "Ikea-AlpsCH_9x6",   # base name of zipfile, .zip will be added
          "CPU_cores_to_use" : 0,  # 0 means all cores, None (null in JSON!) => don't use multiprocessing
          "max_cells_for_memory_only" : 1000 * 1000, # if raster is bigger, use temp_files instead of memory
          "no_bottom": False, # omit bottom triangles?
          #"rot_degs": 0, # rotate by degrees ccw  # CH disabled for now
          "bottom_image": None,  # 1 band greyscale image used for bottom relief
          "ignore_leq": None, # set values <= this to NaN, so they are ignored
          "lower_leq": None,  # e.g. [0.0, 2.0] values <= 0.0 will be lowered by 2mm in the final model
          "unprojected": False, # don't project to UTM, only usefull when using GEE for DEM rasters
          "only": None,# list of tile index [x,y] with is the only tile to be processed. None means process all tiles (index is 1 based)
          "poly_file": None, #"idaho.kml", #TT_poly_test.kml",
          "smooth_borders": True, # smooth borders
```

Any idea what could be causing this? Thanks for your help!

ChHarding, 2023-04-03:
> yeah, it seems like the less tiles i have, the less the severe the misalignment is (which would make sense if its related to the lowest elevation computations)
>
>
> Chris Harding
> Associate Professor
> Department of Geological & Atmospheric Sciences
> Touchterrain.geol.iastate.edu<http://Touchterrain.geol.iastate.edu>

OK, I”m guessing I broke something then recently in that area when I added the min_elev setting. If I get some time tomorrow I’ll have a look.

ChHarding, 2023-04-06:
Luke,

I messed something up during the min_height parameter implementation (as I had suspected) and it should be fixed now. Thanks for telling me!

martinvandriel, 2025-03-22:
Hi Chris,

I still see this issue in the current online tool:

[image]

This is the settings I used: http://touchterrain.geol.iastate.edu/main?trlat=47.80589654394221&trlon=12.627906080560129&bllat=46.8171757058106&bllon=11.100806471185129&scale=1.00&DEM_name=USGS/SRTMGL1_003&tilewidth=200&printres=0.4&ntilesx=2&ntilesy=2&DEMresolution=115.47&basethick=1&zscale=1.0&fileformat=STLb&maptype=roadmap&gamma=1&transp=70&hsazi=315&hselev=45&map_lat=47.659555056153565&map_lon=11.049773151738673&map_zoom=8

Thanks for this great tool in any case
Martin

ChHarding, 2025-04-07:
Quick update:  I did find the core issue for this (each tile was using a slightly different elevation offset) and I seem to have fixed it but I also noticed that there's an (new?) issue for tiles in that their simple 2 triangle bottom is either twice as wide or half as wide, so I'm looking int that now.

________________________________________
From: Martin van Driel
Sent: Saturday, March 22, 2025 6:17 AM
To: ChHarding/TouchTerrain_for_CAGEO
Cc: Harding, Chris [EAC]; Comment
Subject: Re: [ChHarding/TouchTerrain_for_CAGEO] Misaligned Tiles (Issue #77)

Hi Chris,
I still see this issue in the current online tool:
image.png (view on web)
This is the settings I used: http://touchterrain.geol.iastate.edu/main?trlat=47.80589654394221&trlon=12.627906080560129&bllat=46.8171757058106&bllon=11.100806471185129&scale=1.00&DEM_name=USGS/SRTMGL1_003&tilewidth=200&printres=0.4&ntilesx=2&ntilesy=2&DEMresolution=115.47&basethick=1&zscale=1.0&fileformat=STLb&maptype=roadmap&gamma=1&transp=70&hsazi=315&hselev=45&map_lat=47.659555056153565&map_lon=11.049773151738673&map_zoom=8
Thanks for this great tool in any case
Martin
—
Hi Chris,
I still see this issue in the current online tool:
image.png (view on web)
This is the settings I used: http://touchterrain.geol.iastate.edu/main?trlat=47.80589654394221&trlon=12.627906080560129&bllat=46.8171757058106&bllon=11.100806471185129&scale=1.00&DEM_name=USGS/SRTMGL1_003&tilewidth=200&printres=0.4&ntilesx=2&ntilesy=2&DEMresolution=115.47&basethick=1&zscale=1.0&fileformat=STLb&maptype=roadmap&gamma=1&transp=70&hsazi=315&hselev=45&map_lat=47.659555056153565&map_lon=11.049773151738673&map_zoom=8
Thanks for this great tool in any case
Martin

martinvandriel, 2025-05-09:
Any update here? Otherwise I'd hack together some postprocessing to fix the files.

cheers,
Martin

ChHarding, 2025-05-09:
Martin, sorry, I did fix it and never encountered the issue with the bottom I mentioned earlier. So it should be good to go, if not let me know.

martinvandriel, 2025-05-12:
Perfect, thanks! I checked and it works with the example I linked above. Will try the bigger ones I was working on later check in detail.

Cheers,
Martin