ChHarding/TouchTerrain_for_CAGEO #79: 3d STL height differences

jerryhalstead, 2023-07-16:
I'm creating two versions of (ideally) the same terrain stl, except one of them has a gpx file. Even though they have the same settings,  the STLs aren't the same height: the "plain" stl being 47.09391mm while the one with gpx "notch" (using negative gpxPathHeight) ends up taller at 47.45476mm.

What I end up doing with them is bringing into Blender to perform a boolean subtraction that extracts the gpx "trails" (image attached). Then I can combine that with the "notched" terrain file in 3dPrinter slicer to print in separate colors.

I can get it working, with some manual tweaking and post processing of the STLs, just thought I'd report the issue in case it's something that can be fixed or controlled (like passing in a max height in mm?).

Or, even better, an option to output two STLs: the base "notched" terrain and the part that was removed (i.e. the trail). That would be sweet. 

[logfile.txt](https://github.com/ChHarding/TouchTerrain_for_CAGEO/files/12064749/logfile.txt)
[logfile.txt](https://github.com/ChHarding/TouchTerrain_for_CAGEO/files/12064750/logfile.txt)
[image]

Maxvi33, 2023-12-27:
Hello,
You succeed ? 
I'l trying to do exactly same thing to print in 3d tbut substraction boolen in fusion 360 its hard with stl.. and the difference of heights generate dont help me :)

PS:  your gpx looks good, me, i lost in resolution :(

ChHarding, 2023-12-29:
Hi,

There’s no solution to create STLs of GPX trails directly in TouchTerrain at the moment. However, I am working on a somewhat similar enhancement that could also make your request possible. I’ll let you know if/when that happens.

Cheers

Chris

Chris Harding
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu<http://Touchterrain.geol.iastate.edu>

Maxvi33, 2023-12-30:
Hello, thanks for your answer & help 🙂
So good if you are working on an alternative. I Can help if needed !

For my part, i succeeded a little:
[image]
(moreover, with the same DEM and parameters, on the standalone script (touchterrain_standalone.py) i generate an entire tile but with Jupyter(TouchTerrain_standalone_jupyter_notebook.ipynb), I generate the coast cut out as above)

But the ideal +++ would be to be able to generate a .step to use it directly in CAD software 🥹

A bientôt 
Max

ChHarding, 2024-01-01:
Hello, thanks for your answer & help 🙂
So good if you are working on an alternative. I Can help if needed !

Currently testing my current issue, after that I’ll work on this. I don’t think it’s too tricky.

For my part, i succeeded a little:
IMG-20231229-WA0000.jpg (view on web)<https://github.com/ChHarding/TouchTerrain_for_CAGEO/assets/7173062/b56a0763-79b9-489a-9ee2-5b55c89931bb>
(moreover, with the same DEM and parameters, on the standalone script (touchterrain_standalone.py) i generate an entire tile but with Jupyter(TouchTerrain_standalone_jupyter_notebook.ipynb), I generate the coast cut out as above)

That looks very nice!

But the ideal +++ would be to be able to generate a .step to use it directly in CAD software 🥹

That's not going to happen, STEP files (AFAIK) are for parametric objects not for meshes. But can’t you import STL into Fusion?

A bientôt
Max

—

Chris Harding
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu<http://Touchterrain.geol.iastate.edu>