ChHarding/TouchTerrain_for_CAGEO #81: STL top max value not scaling correctly

TrainerRed151, 2023-12-06:
I'm running the standalone Jupyter notebook version inside of the docker container.  I downloaded 2 (because the mountain I'm interested in, Pikes Peak, falls right on the boundary) GeoTIFF files on the USGS website from the 3DEP program, merged them using QGIS, and then extracted just the area I wanted into a new GeoTIFF file.  When I run the python code, it seems to mess up the scaling and the top max value for the model ends up being like 2000m.  So the resulting dimensions of the STL object end up as 300mm x 300mm x 2000000mm.  Looking at the log files from other issues submitted here, it seems like the problem might be that the code is interpreting the lat/long values of the raster file as x/y values, hence messing up the scaling.

```
Log for creating 1 x 1 3D model tile(s) from pikes_peak_area.tif 
 
started: 20:17:59.980281 
Warning: raster cells are not square (9.259300388672895e-05x9.259341472438034e-05) , using9.259300388672895e-05
source raster upper left corner (x/y):  -106.00056 39.00056 
source raster cells size 9.259300388672895e-05 m  (10812, 21612) 
projection: WGS 84 
z-scale: 1 
basethickness: 0.5 
fileformat: STLb 
tile_centered: False 
no_bottom: True 
no_normals: True 
ignore_leq: None 
lower_leq: None 
importedGPX: ['stuff/Pikes_Peak_via_Barr_Trail.gpx'] 
undefined GDAL value: -3.4028234663852886e+38 
process gpx file: stuff/Pikes_Peak_via_Barr_Trail.gpx 
1 GPX tracks found 
Plotting track 1 
Time to add GPX paths:1.0155231952667236 
tile_width: 300 
tile_height: 150.08328706274293 
source raster width 2.001119999999986 m, cell size: 9.259300388672895e-05 m, elev. min/max is -5.369226455688477 65753.40625 m 
source raster 3D print resolution would be 0.0138811771238201 mm 
re-sampling scaled.tif :
  (21612, 10812) 0.0138811771238201 mm  9.259300388672895e-05 m  -5.369226455688477 - 65753.40625 m to 
  (1499, 750) 0.200133422281521 mm  0.0013340799999999908 m  0.0009282126 - 65534.996 m 
after resampling, requested print res was adjusted from 0.2 to 0.200133422281521 to ensure correct model dimensions 
map scale is 1 : 6.665953066666621 
Cells per tile (x/y) 1499 x 750 
using single-core only (multi-core is currently broken :( 

1 x 1 tiles, tile size 300.00 x 150.10 mm
 
tile 1 1 : height:  0.0009282126 - 9831302.0 mm , file size: 108 Mb 

total size for all tiles: 108 Mb 

processing finished: 20:18:23.266631 
```

ChHarding, 2023-12-07:
If I understand you correctly you’re using a geotiff that uses an unprojected coordinate system. That’s really not advisable precisely b/c the x/y units (degrees) are different from z (meters or ft). What you should do is go into QGIS and save your file with a UTM projection. Right click on the final raster layer and choose Export - Save as.  I

http://tmackinnon.com/wp-content/uploads/utm-USA.jpg. should tell you which zone to use (11 or 12 or whatever). Once you know the UTM zone (X), click on the globe thingy on the right and type in UTM X in the filter. In the list below choose WGS84 / UTM zone X N (for North) and hit OK. (WGS84 is a global datum). Give the new raster a filename (????_UTMX.tif) and save the new raster  (no need to change anything else, it should change the resolution shown into meters).  That should work, if not, report back.

Cheers

Chris

Chris Harding
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu<http://Touchterrain.geol.iastate.edu>

TrainerRed151, 2023-12-09:
Assigning a projection to the tif file fixed the problem, thanks!

closed 2023-12-09 by TrainerRed151