ChHarding/TouchTerrain_for_CAGEO #86: STL not watertight

ThomasPe, 2023-12-25:
I noticed that the STL created by Touchterrain isn't watertight, therefore giving me errors in most tools I import the model. The tools are usually able to fix the issue (e.g. Microsoft 3D Builder, Fusion 360) but having a correct STL returned would obviously be better.

This can also be confirmed programatically using trimesh:
 ```
import trimesh
data = trimesh.load('/workspaces/map2stl/test/V2_2_11.18_48.01_tile_1_1.STL')
print(data .is_watertight)
```

I started looking into why that is but wanted to track it here early in case this is a known issue.

ChHarding, 2024-01-01:
Noted!  This is something I’ve tried to fix for a long time and never found a good solution. My assumption is that the coordinates that end up in to STL binary file get truncated from the float64 I use for processing into float32 which is what the STL spec requires. My only workaround for this is to use ASCII STL, so this would be my first suggestion to try with your case.  The other, unsolvable issue are non-manifold vertices which get created when 2 cells touch diagonally (0 means undefined here, 1 is a valid elevation) like this:

01  10
10  01

I don’t think that’s a watertight issue but if it is, use clean_diags=True which will remove on of the two by setting it to undefined. The issue is also helped by using smooth_borders=True as that snips off one of the two triangles each cell gets and so lessens that chance for the diagonal case happening.

Unrelated:  I’m a bit stuck getting your devcontainer stuff to start. I have the DevContainer extension for VSCode and haveDockerDesktop and WSL installed bit I just can’t figure out how to get the magic started :) I expect I need to do something with the 2 files in .devcontainer?  How do I say: run the devcontainer.json file?

Cheers

Chris

Chris Harding
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu<http://Touchterrain.geol.iastate.edu>

ThomasPe, 2024-01-01:
thanks, I'll give those options a shot.
regarding the dev container, here's a screenshot of the prompt I get when opening the project in vs code:

[image]

you can also click the bottom left corner in the editor to start the dev container:
[image]

ThomasPe, 2024-01-01:
can confirm that setting `clean_diags` to true solved the issue. Trimesh reports as watertight and both Fusion 360 and Microsoft 3D Builder don't need to repair it. Feels like this should be set to true as default. :D

closed 2024-01-01 by ThomasPe

ChHarding, 2024-01-01:
Cool!  Well, the processing can be kinda slow b/c it has to usually do it multiple times until all issues are fixed. I think I would rather use a better explanation.

BTW I got the devcontainer to work but only when I use Clone in Volume, Reopen in Container gave me lots of error. One silly question: if I run it in the cloned volume, how do I copy a file into it? I have some config files that I need to copy.

ThomasPe, 2024-01-02:
I see, this seems to be what you are working with:
https://code.visualstudio.com/docs/devcontainers/containers#_quick-start-open-a-git-repository-or-github-pr-in-an-isolated-container-volume

on the second try I was able to clone and mount it as you did. For me, simply CTRL-C - CTRL-V copy-pasting works from the Explorer to VS Code.

ChHarding, 2024-01-02:
Oh yeah, copy/paste from Explorer works (duh!), thanks :) So yeah, this is really impressive, thanks for pointing me to it. Do you think it’s worth writing this up here: https://github.com/ChHarding/TouchTerrain_for_CAGEO/tree/master#standalone-mode as an alternative to windows? I guess I don’t have a good idea how many people are sufficiently savvy with Docker and WSL …

Cheers

Chris

https://code.visualstudio.com/docs/devcontainers/containers#_quick-start-open-a-git-repository-or-github-pr-in-an-isolated-container-volume

ChHarding, 2024-01-02:
One more thing: thinking about working with the docker container, I would find it useful to be able to mount my Docker project root to some windows folder b/c in development I often want to run windows apps on processing results (STLs or tiffs)

I found this: https://docs.docker.com/guides/walkthroughs/access-local-folder/ which uses a yaml file to mount a local folder to a container folder but I don’t know enough about Docker to determine how I would do the same, specifically b/c there’s no yaml file in ,devcontainer  Should I try to mimic the yaml volumes setting in the devcontainer.josn file that I have? Or just create a new compose.yaml file?

https://code.visualstudio.com/docs/devcontainers/containers#_quick-start-open-a-git-repository-or-github-pr-in-an-isolated-container-volume

ChHarding, 2024-01-03:
Thomas,

I just run a really simple test for the use_geo_coords=UTM case and it seems to work. I can only use ArcGIS for this, so I created a random elevation raster at a location (using UTM), run it through TouchTerrain, saved it as obj (ArcGIS can’t import STL) and imported it into ArcGIS, As far as I can tell it overlaps quite well:

Here are some screenshots, my geotif,  the processing parameters and my resulting files: https://drive.google.com/drive/folders/1yDGXUCQRCs37WHdd5lEgMNK_fsuyP3b5?usp=sharing

FYI, the crux seems to be to set printers to -1 (use raster resolution). For now it won’t work for anything else e.g. 0,4 mm. Also, umm will ignore tilewidth and tilewidth_scale.

Let me know if this works for you and we can talk what else I need to change/fix.

Cheers

Chris

https://code.visualstudio.com/docs/devcontainers/containers#_quick-start-open-a-git-repository-or-github-pr-in-an-isolated-container-volume