ChHarding/TouchTerrain_for_CAGEO #88: Preview .STL file in notebook while running in Docker

86cucv, 2024-02-03:
In the notebook `TouchTerrain_jupyter_for_starters.ipynb` it mentions the following in the **Preview the model** section: 

> Dec. 8, 2022: This may not work when run inside Docker ...

I found that all a user needs to do is run the following commands within the docker container before launching the notebook:

`jupyter nbextension install --py --user k3d
jupyter nbextension enable --py --user k3d`

Source: https://github.com/K3D-tools/K3D-jupyter/issues/324#issuecomment-1004784211

ChHarding, 2024-02-05:
Very cool! Thank you!

Source: K3D-tools/K3D-jupyter#324 (comment)<https://github.com/K3D-tools/K3D-jupyter/issues/324#issuecomment-1004784211>

Chris Harding
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu<http://Touchterrain.geol.iastate.edu>

ChHarding, 2024-02-23:
Hi,

I just added the jupyer 3D extension install and enable to  TouchTerrain_jupyter_for_starters.ipynb. I also tried to make this work for the Colab, which I think is the most common way for docker newbs to run the standalone version, but I’ve not gotten this to work. I ask K3D-jupyter https://github.com/K3D-tools/K3D-jupyter/issues/324#issuecomment-1932642817  and go some feedback but there’s still an issue and for that I didn’t get an answer :( I think it has to do with ipywidgets which I’m not at all familiar with. Are you by any chance familiar with it and could have a look at what I have so far? If so, I’ll send you my colab notebook.

Cheers

Chris