ChHarding/TouchTerrain_for_CAGEO #89: Website is down

DzmitryFil, 2024-03-06:
Not sure if this is the place for this issue, but websites linked in the readme do not work anymore.
[image]

ChHarding, 2024-03-06:
Happy to fix this but I need to know which URL you were trying to use(?)

image.png (view on web)<https://github.com/ChHarding/TouchTerrain_for_CAGEO/assets/5418032/b8776b56-3b86-40a5-8934-6f8b2babe701>

Chris Harding
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu<http://Touchterrain.geol.iastate.edu>

ChHarding, 2024-03-06:
I just realized you were talking about the touchterrain web app! It’s up again now.

closed 2024-03-06 by DzmitryFil