ChHarding/TouchTerrain_for_CAGEO #90: SYNTAX FOR USING "importedGPX" :

mandeep255, 2024-03-22:
I am unable to pass a path for a valid file, I am using syntax  "importedGPX" :  "C:\\v.gpx" 

I get an error message 

importedGPX = C:\v.gpx

Error: [Errno 2] No such file or directory: 'C'

can you help pl, thanks

mandeep255, 2024-03-24:
I tried with the following parameters separately
"importedGPX": “C:\\v.gpx”
"importedGPX": r“C:\v.gpx”

I am getting the following messages in each case

Warning: JSON decode Error for manual: "importedGPX": “C:\\v.gpx” Expecting value: line 1 column 18 (char 17) (ignored)
Warning: JSON decode Error for manual: "importedGPX": r“C:\v.gpx” Expecting value: line 1 column 18 (char 17) (ignored)

I must be doing something wrong!
regards

ChHarding, 2024-03-25:
It’ll take some time before I can dig into the code. For now, how about this:

- copy the gpu file into the same folder that the python file your are running is in (I assume you’re running TouchTerracin_standalone.py, right?)
- change your config to "importedGPX": “v.gpx”. And see if that works

Chris Harding
Associate Professor
Department of Geological & Atmospheric Sciences
Touchterrain.geol.iastate.edu<http://Touchterrain.geol.iastate.edu>

mandeep255, 2024-03-25:
I am using the web version!

ChHarding, 2024-03-25:
OK :). Well, sorry, that won’t work b/c I’ve not put any any hooks in for an upload to the browser for gpx files (like I did for uploading kms files).  You could try the Colab version of the jupyter notebook:

In a browser where you are logged in with you Google account go to this:  http://colab.research.google.com/github/ChHarding/TouchTerrain_for_CAGEO/blob/master/TouchTerrain_jupyter_starters_colab.ipynb
There’s an example of how to work with a gpx file (using a file in the stuff folder) so you could upload your file there and just change the name in the notebook cell accordingly.