ChHarding/TouchTerrain_for_CAGEO #95: adding gpx route failed

yuvalorp, 2024-07-22:
i am trying to add gpx to the model with the standalone docker container and get the issue
[image]

ChHarding, 2024-07-22:
The gpx stuff isn’t really mine but I’m guessing the errors have to do with the projection of the gpx data (which are unprojected lat/long points with WGS84) into the coordinate system of the DEM raster.  Does it say anywhere what the coordinate system of your Geotiff is?

image.png (view on web)<https://github.com/user-attachments/assets/00fd4f96-4a81-441a-895f-54d770bd7204>

yuvalorp, 2024-08-05:
thanks for the response :) i am taking the Geotiff from AW3D30 , in the internet i saw that it used a EGM96 projection (in the paper is written "Elevation (in meter) converted from the ellipsoidal height based on
ITRF97 and GRS80, using EGM96†1 geoid model")

ChHarding, 2024-08-06:
Well I’m guessing that that’s a combination that gdal can’t convert to lat/long. The only solution I can think of is to convert the coordinate system to something more common, like the UTM zone it’s inside and WGS84 as datum. If you tell me where your area is I can figure out the UTMs zone and tell you how to convert to in in QGIS.  Or, if this is a one-shot, if you give me access to the file I can convert it for you.