ChHarding/TouchTerrain_for_CAGEO #96: `"tile_centered": true` has no effect

KIC, 2024-08-17:
The `python TouchTerrain_standalone.py` returns the same result whether `tile_centered` is true or not. Since I want to postprocess the generated tiles in openSCAD I actually would like all tiles to be centered.

ChHarding, 2024-08-17:
Interesting, I'll look into it. May I ask what you're using the tiles for?

KIC, 2024-08-17:
I just want to add some custom connectors instead of straight walls.

ChHarding, 2024-08-17:
Well, I just a simple 2 x 2 tiles case with tile_centered as true and as false and both cases seem to work (see zipped images, hopefully this works with git mail ...). 
This is the json file I used:

{
"CPU_cores_to_use": null,
"DEM_name": "USGS/3DEP/10m",
"basethick": 0.6,
"bllat": 39.32205105794382,
"bllon": -120.37497608519418,
"bottom_elevation": null,
"bottom_image": null,
"clean_diags": false,
"fileformat": "STLb",
"fill_holes": null,
"ignore_leq": null,
"importedDEM": "top/WV_top_cl6.tif",
"importedGPX": [],
"lower_leq": null,
"max_cells_for_memory_only": 25000000,
"min_elev": null,
"no_bottom": false,
"ntilesx": 2,
"ntilesy": 2,
"offset_masks_lower": null,
"only": null,
"poly_file": null,
"printres": 0.4,
"smooth_borders": true,
"tile_centered": true,
"tilewidth": 120,
"tilewidth_scale": null,
"trlat": 39.45763749030933,
"trlon": -120.2002248034559,
"unprojected": false,
"zip_file_name": "tilecenteredviajson",
"zscale": 2.0
}

I also tried the same data using the override settings inside the python file instead of a json file. So can you show me what settings your are using and how "tile_centered": true does not work for you?

(Note: I admit that I've been working on the part of code around where the tile centering is located for several month now, implementing a new feature and although I guess its possible that I accidentally fixed something, I'm pretty sure I've had no reason to touch it. But, I've been know to not see simple things, so I could be wrong ...)

Cheers

Chris

KIC, 2024-08-17:
Hmm ... when I use this json with the standalone py
```
{
"CPU_cores_to_use": 0,
"DEM_name": "JAXA/ALOS/AW3D30/V2_2",
"basethick": 1,
"bottom_image": null,
"clean_diags": false,
"fileformat": "STLb",
"fill_holes": null,
"gpxPathHeight": 100,
"gpxPathThickness": 5,
"gpxPixelsBetweenPoints": 20,
"ignore_leq": null,
"importedDEM": null,
"importedGPX": null,
"lower_leq": null,
"max_cells_for_memory_only": 1000000,
"min_elev": null,
"no_bottom": true,
"no_normals": true,
"ntilesx": 4,
"ntilesy": 4,
"offset_masks_lower": null,
"only": null,
"polyURL": null,
"poly_file": null,
"printres": 0.2,
"smooth_borders": false,
"tile_centered": true,
"tilewidth": 100,
"tilewidth_scale": null,
"trlat": 28.59611007775015,
"trlon": -16.10975691796874,
"bllat": 27.9878751235126,
"bllon": -16.932816000000003,
"unprojected": false,
"use_geo_coords": "centered",
"zip_file_name": "tenerife",
"zscale": 1.0
}
```

and incluide them in openscad:

```import("/tmp/tt/V2_2_-16.52_28.29_tile_1_1.STL");
import("/tmp/tt/V2_2_-16.52_28.29_tile_1_2.STL");
import("/tmp/tt/V2_2_-16.52_28.29_tile_1_3.STL");
import("/tmp/tt/V2_2_-16.52_28.29_tile_1_4.STL");
```

they are not overlapping as i would expect:

[image]

ChHarding, 2024-08-17:
I see! I'll test this but I suspect that the issue has to do with "use_geo_coords": "centered"  Is there a pressing reason why you don't want you models to be in mm? Is this going into some sort of GIS system where units in meters (i.e. UTM) instead of mm is needed? If not try not using it (or setting it to null) and see if that solves your issue.

image.png (view on web)<https://github.com/user-attachments/assets/8874ec30-2472-4f3e-8508-28e2f86c6da8>

KIC, 2024-08-18:
actually I used "use_geo_coords": "centered" because I hoped it will be a solution. but the default null settings look very similar, just the whole island is not in the center. I really only want each individual tile to be centered.