ChHarding/TouchTerrain_for_CAGEO #102: Web app preview not loading

kathleenhanke, 2025-03-05:
Preview not loading after export with web app. Not sure how else to describe it. 
It seems I'm able to download the STL fine but the preview never seems to load. It was working a few weeks ago but now only a dashed lined box appears when previously I could see the model and turn it.

Any help would be appreciated.  My students have loved using this page. 

[image]

ChHarding, 2025-03-07:
We had to change servers a couple of weeks ago and I guess that broke it. Thanks for the note, I'll look into it.

Screenshot.2025-03-04.8.26.04.PM.png (view on web)<https://github.com/user-attachments/assets/d513992a-4566-4425-aa5c-1c07302d9a53>

Screenshot.2025-03-04.8.26.04.PM.png (view on web)<https://github.com/user-attachments/assets/d513992a-4566-4425-aa5c-1c07302d9a53>

kathleenhanke, 2025-03-10:
> Thanks for the note, I&#39;ll look into it.<br> <br>
> ________________________________________<br> From: kathleenhanke
> ChHarding/TouchTerrain_for_CAGEO<br> Cc: Subscribed<br> Subject:
> #102)<br> <br> Preview not loading after export with web app. Not sure how
> else to describe it.<br> It seems I&#39;m able to download the STL fine but
> turn it.<br> <br> Any help would be appreciated. My students have loved
> using this page.<br> <br> Screenshot.2025-03-04.8.26.04.PM.png (view on
> web)&lt;
> unsubscribe&lt;
> (ChHarding/TouchTerrain_for_CAGEO#102)&lt;
> <br> Preview not loading after export with web app. Not sure how else to
> describe it.<br> It seems I&#39;m able to download the STL fine but the
> it.<br> <br> Any help would be appreciated. My students have loved using

THANK YOU!
We look forward to being able to preview again!

Kathleen Hanke
Apple Teacher
Den Ambassador
Google Certified Educator Level 1
Google Certified Educator Level 2
Google for Education Certified Trainer
Innovation Specialist Parker High School
Cell (608) 314-5863
Google Voice- (608) 713-1199

*Get Tech Help Now *

   - Call 608-743-5004 to speak with a technician
   - Fill out a *Help Ticket
   <https://schooldistrictofjanesville.samanage.com/login>(choose New Ticket)*

ChHarding, 2025-05-07:
It's fixed now.

________________________________________
From: Harding, Chris [EAC]
Sent: Friday, March 7, 2025 12:18 PM
To: ChHarding/TouchTerrain_for_CAGEO
Subject: Re: [ChHarding/TouchTerrain_for_CAGEO] Web app preview not loading (Issue #102)

We had to change servers a couple of weeks ago and I guess that broke it. Thanks for the note, I'll look into it.