ChHarding/TouchTerrain_for_CAGEO #103: Search page errors

Alissa1234568, 2025-04-15:
the main page seems to be malfunctioning. You can't search for locations or put a box around the desired location. Most of the main functions on the homepage seem to be malfunctioning.

ChHarding, 2025-04-16:
Thanks for the note. Seems there were a  couple of recent API changes that broke a few things. Working on it it not sure how long it'll be. 

________________________________________
From: Alissa1234568
Sent: Tuesday, April 15, 2025 3:21 PM
To: ChHarding/TouchTerrain_for_CAGEO
Cc: Subscribed
Subject: [ChHarding/TouchTerrain_for_CAGEO] Search page errors (Issue #103)

the main page seems to be malfunctioning. You can't search for locations or put a box around the desired location. Most of the main functions on the homepage seem to be malfunctioning.

Alissa1234568, 2025-04-16:
ok, sounds good. Thank you so much for getting back to me :)

ChHarding, 2025-04-17:
It's back to working again. Can I ask what you are using TouchTerrain for?

________________________________________
From: Alissa1234568
Sent: Wednesday, April 16, 2025 6:30 PM
To: ChHarding/TouchTerrain_for_CAGEO
Cc: Harding, Chris [EAC]; Comment
Subject: Re: [ChHarding/TouchTerrain_for_CAGEO] Search page errors (Issue #103)

ok, sounds good. Thank you so much for getting back to me :)

Alissa1234568, 2025-04-17:
YAY! thank you so much. I'm a Jeweler. I make topographic jewelry. I call them my wearable mountains, haha.
 Chaparral Jewelry :)

ChHarding, 2025-04-21:
Wow that's cool :)

________________________________________
From: Alissa1234568
Sent: Thursday, April 17, 2025 6:17 PM
To: ChHarding/TouchTerrain_for_CAGEO
Cc: Harding, Chris [EAC]; Comment
Subject: Re: [ChHarding/TouchTerrain_for_CAGEO] Search page errors (Issue #103)

YAY! thank you so much. I'm a Jeweler. I make topographic jewelry. I call them my wearable mountains, haha.Chaparral Jewelry :)

ChHarding, 2025-10-22:
> Hello! I was hoping to bring the error message on TouchTerrain to your attention. I apologize in advance if you're already working on the issue. Thank you so much for maintaining this amazing tool.

Thanks for the note, it's working again.