ChHarding/TouchTerrain_for_CAGEO #104: tif is not included in GeoTiff format

sdobz, 2025-05-01:
I prefer to do my own generation of geometry so I have previously used the geotif export feature.

Initially I was using a jupyter notebook packaged using [nix](https://github.com/sdobz/jupyter-gis) BTW: this was a LOT of work, but ooo boy it'll work for a long time.

After setting
```python
args["fileformat"] = "GeoTiff"
```

I got a zip file, but it was mostly empty.
```
Using GeoJSON polygon for masking with 5 points
Log for creating 3D model tile(s) for  10m_-121.69_45.38

DEM_name = USGS/3DEP/10m
trlat = 45.486506690000006
trlon = -121.51919176999999
bllat = 45.271522309999995
bllon = -121.85887523000001
printres = 0.4
ntilesx = 1
ntilesy = 1
tilewidth = 120
basethick = 0
zscale = 2.0
fileformat = GeoTiff
no_bottom = False
unprojected = False
no_normals = True

process started: 17:07:42.025898

Region (lat/lon):
   45.486506690000006 -121.51919176999999 (top right)
   45.271522309999995 -121.85887523000001 (bottom left)
center at [-121.6890335, 45.3790145], UTM10N, EPSG:32610
lon/lat size in degrees: [0.3396834600000176, 0.2149843800000113]
Earth Engine raster: USGS/3DEP/10m
URL for geotiff is:  https://earthengine.googleapis.com/v1/projects/wax-cam/thumbnails/3ae10a9001a8890e4b9c6498c748ff42-5c2fcd0a77f766fb92aed58a251d3624:getPixels
 geotiff size: 0.25784778594970703 Mb
 cell size 88.6859716835138 m, upper left corner (x/y):  589406.9678086328 5037895.3074536845

processing finished: 17:07:43.689343
```

The webapp has the same issue: this url previously included a tif, now it is a zip with just the logfile.

http://touchterrain.geol.iastate.edu/main?trlat=45.59897980851287&trlon=-121.377448046875&bllat=45.14652837026163&bllon=-122.02838798828125&scale=1.00&DEM_name=USGS/3DEP/10m&tilewidth=300&printres=0.199&ntilesx=1&ntilesy=1&DEMresolution=33.82&basethick=1&zscale=1.0&fileformat=GeoTiff&maptype=roadmap&gamma=1&transp=20&hsazi=315&hselev=45&map_lat=45.39629585667898&map_lon=-121.75373699315436&map_zoom=10

The source code offers a clue
https://github.com/ChHarding/TouchTerrain_for_CAGEO/blob/082548b160b8a90d47798b1d15b858407ad876f3/touchterrain/common/TouchTerrainEarthEngine.py#L1874

This line was changed 2 months ago:
https://github.com/ChHarding/TouchTerrain_for_CAGEO/commit/3af0e0bd129318c1fbc9bfb1e45013114dadac81

Unsure what the intent is there, but this probably isn't what you want!

I'd love to be able to compose parts of the code together, specifically I am analyzing the heightmap to center a circle on the peak and determine the minimum elevation in a ring around the center point.
Three options:
1. Run touchterrain to get a tif, analyze update and generate a dem, then run it again on the importedDEM
2. Ruthlessly copy paste bits of code out of this project and into my own
3. Refactor `TouchTerrainEarthEngine` to have composable elements

For now I'm probably going to fix this bug in a fork and go with 1, thanks for this excellent project! I love the mission of tactile terrain and have been investing considerable time into it.

sdobz, 2025-05-01:
Here's some projects, pardon the irrelevancy I'm just stoked to share 😁 

[image]
[image]

ChHarding, 2025-05-01:
Vincent,

I'm aware of geotiff export not getting a plot/histgram file. the "fix" a while ago was needed to make geotiff export working again but I'm happy to look at it again. 

Cheers

Chris 

________________________________________
From: Vincent Khougaz
Sent: Wednesday, April 30, 2025 7:48 PM
To: ChHarding/TouchTerrain_for_CAGEO
Cc: Subscribed
Subject: [ChHarding/TouchTerrain_for_CAGEO] tif is not included in GeoTiff format (Issue #104)

[X]sdobz created an issue (ChHarding/TouchTerrain_for_CAGEO#104)I prefer to do my own generation of geometry so I have previously used the geotif export feature.Initially I was using a jupyter notebook packaged using nix BTW: this was a LOT of work, but ooo boy it'll work for a long time.After settingargs["fileformat"] = "GeoTiff"I got a zip file, but it was mostly empty.Using GeoJSON polygon for masking with 5 points
Log for creating 3D model tile(s) for  10m_-121.69_45.38

DEM_name = USGS/3DEP/10m
trlat = 45.486506690000006
trlon = -121.51919176999999
bllat = 45.271522309999995
bllon = -121.85887523000001
printres = 0.4
ntilesx = 1
ntilesy = 1
tilewidth = 120
basethick = 0
zscale = 2.0
fileformat = GeoTiff
no_bottom = False
unprojected = False
no_normals = True

process started: 17:07:42.025898

Region (lat/lon):
   45.486506690000006 -121.51919176999999 (top right)
   45.271522309999995 -121.85887523000001 (bottom left)
center at [-121.6890335, 45.3790145], UTM10N, EPSG:32610
lon/lat size in degrees: [0.3396834600000176, 0.2149843800000113]
Earth Engine raster: USGS/3DEP/10m
URL for geotiff is:  https://earthengine.googleapis.com/v1/projects/wax-cam/thumbnails/3ae10a9001a8890e4b9c6498c748ff42-5c2fcd0a77f766fb92aed58a251d3624:getPixels
 geotiff size: 0.25784778594970703 Mb
 cell size 88.6859716835138 m, upper left corner (x/y):  589406.9678086328 5037895.3074536845

processing finished: 17:07:43.689343
The webapp has the same issue: this url previously included a tif, now it is a zip with just the logfile.http://touchterrain.geol.iastate.edu/main?trlat=45.59897980851287&trlon=-121.377448046875&bllat=45.14652837026163&bllon=-122.02838798828125&scale=1.00&DEM_name=USGS/3DEP/10m&tilewidth=300&printres=0.199&ntilesx=1&ntilesy=1&DEMresolution=33.82&basethick=1&zscale=1.0&fileformat=GeoTiff&maptype=roadmap&gamma=1&transp=20&hsazi=315&hselev=45&map_lat=45.39629585667898&map_lon=-121.75373699315436&map_zoom=10The source code offers a cluehttps://github.com/ChHarding/TouchTerrain_for_CAGEO/blob/082548b160b8a90d47798b1d15b858407ad876f3/touchterrain/common/TouchTerrainEarthEngine.py#L1874This line was changed 2 months ago:3af0e0bUnsure what the intent is there, but this probably isn't what you want!I'd love to be able to compose parts of the code together, specifically I am analyzing the heightmap to center a circle on the peak and determine the minimum elevation in a ring around the center point.Three options:Run touchterrain to get a tif, analyze update and generate a dem, then run it again on the importedDEMRuthlessly copy paste bits of code out of this project and into my ownRefactor TouchTerrainEarthEngine to have composable elementsFor now I'm probably going to fix this bug in a fork and go with 1, thanks for this excellent project! I love the mission of tactile terrain and have been investing considerable time into it.

ChHarding, 2025-05-01:
Very nice, never would have though of that :)

________________________________________
From: Vincent Khougaz
Sent: Wednesday, April 30, 2025 7:49 PM
To: ChHarding/TouchTerrain_for_CAGEO
Cc: Subscribed
Subject: Re: [ChHarding/TouchTerrain_for_CAGEO] tif is not included in GeoTiff format (Issue #104)

[X]sdobz left a comment (ChHarding/TouchTerrain_for_CAGEO#104)Here's some projects, pardon the irrelevancy I'm just stoked to share 😁P1060860.JPG (view on web)PXL_20250428_032906824.jpg (view on web)

ChHarding, 2025-05-02:
Vincent, I think I fixed it but I would like you to test it first on beta before I push it to prod. Can you give https://touchterrain-beta.geol.iastate.edu/  a bit of a workout?

sdobz, 2025-05-02:
I was able to get a tif out of my previous url, but my attempts to get a stl are failing with timeout